boltgolt/howdy #760: ImportError: /usr/lib/libopencv_viz.so.407: undefined symbol: _ZN12vtkAlgorithm8RegisterEP13vtkObjectBase

Integral-Tech, 2023-02-28:
╰─ sudo howdy test    
Traceback (most recent call last):
  File "/usr/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 9, in <module>
    import cv2
ImportError: /usr/lib/libopencv_viz.so.407: undefined symbol: _ZN12vtkAlgorithm8RegisterEP13vtkObjectBase

╰─ sudo howdy snapshot
Traceback (most recent call last):
  File "/usr/bin/howdy", line 107, in <module>
    import cli.snap
  File "/usr/lib/security/howdy/cli/snap.py", line 7, in <module>
    import snapshot
  File "/usr/lib/security/howdy/snapshot.py", line 4, in <module>
    import cv2
ImportError: /usr/lib/libopencv_viz.so.407: undefined symbol: _ZN12vtkAlgorithm8RegisterEP13vtkObjectBase
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Arch Linux

Howdy version (`sudo howdy version`): 2.6.1

Integral-Tech, 2023-02-28:
> Yep, me too, seems to be after an update this morning. A roll back fixes it until the update is run again then it beaks again.

I also tried howdy 3.0.0 beta but in vain :(

Hamster1962, 2023-02-28:
I'm not sure how much you know about Howdy but these are the things that get updated and break it.
[image]
Some are clearly not involved but others..........
H

boltgolt, 2023-02-28:
Vtk is in the missing symbol name, maybe https://stackoverflow.com/questions/56150527/i-am-getting-linking-error-while-compilong-opencv4-basic-program/56152596 has the fix?

Jellyfrog, 2023-02-28:
its the new version of vtk that caused it.
Downgrading it like so works; `pacman -U file:///var/cache/pacman/pkg/vtk-9.1.0-26-x86_64.pkg.tar.zst`

Hamster1962, 2023-02-28:
Thanks @boltgolt & @Jellyfrog. That was it.
I've added an ignore in pacman.conf for vtk.
Is this a permanent fix or is there a better way?

Martinits, 2023-03-02:
I'm on archlinux with howdy 2.6.1 and downgraded vtk, too. But there's still error on `sd howdy test`:
```
Traceback (most recent call last):
  File "/usr/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 9, in <module>
    import cv2
ImportError: /usr/lib/libopencv_viz.so.407: undefined symbol: _ZNK13vtkObjectBase20GetObjectDescriptionB5cxx11Ev
```

Hamster1962, 2023-03-02:
Two more updates today caused issues. Opencv and Python-opencv. Both added as ignore in pacman, but
We should really try and fix this as ignoring updates to Python and/or opencv is not a long term fix.

Integral-Tech, 2023-03-03:
After the recent upgrade, the problem was fixed.

closed 2023-03-03 by Integral-Tech

Hamster1962, 2023-03-03:
OK, thanks, so should i allow the updates?