boltgolt/howdy #763: howdy while lid is closed

rogelioamancisidor, 2023-03-07:
I have installed howdy and works perfectly. Most of the times, I work using external monitor and camera. If I specified the `device_path` to the external usb camera, `howdy` works fine using the usb external camera as long as the lid is opened (even though the laptop's monitor is off). 

How can I change this behavior and use the usb camera but with the lid closed? I have seen this question #515, which I think it suggests that it is possible to use `howdy` with external usb camera and closed lid. BTW, `HandleLidSwitch` is set to `ignore` in my `logind.conf` file.

rogelioamancisidor, 2023-03-07:
I found out the option `ignore_closed_lid` in the configuration file. Setting it to `false` solves my problem.

closed 2023-03-07 by rogelioamancisidor

whowantsmybigdata, 2024-01-31:
added to wiki