boltgolt/howdy #768: pam_howdy freezes and causes calling programs to freeze

fufexan, 2023-03-23:
When I try to authenticate through Howdy and it fails, the calling program (greetd or swaylock) freezes.
I originally thought it was a greetd issue, and opened an issue [here](https://todo.sr.ht/~kennylevinsen/greetd/41#event-230469). As discussed there, it turns out it is actually a `pam_howdy` bug.

- [x] I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): NixOS Stoat 23.05

Howdy version (`sudo howdy version`): `3.0.0 BETA`

Catty2014, 2023-06-01:
Same issue here. `pam_howdy` causes sudo and KDE lock screen to crash.
It seems that pam_howdy threw an exception, making lockers to crash.

SimonBrandner, 2023-07-26:
I can also repro and it's a showstopper for me since I don't want my KDE lock screen crashing on me if it can't detect me :/

SimonBrandner, 2023-07-29:
I don't seem to be able to repro on `beta` anymore - can anyone reproduce? If not, I suggest closing this issue

fufexan, 2023-07-29:
Right, it seems to work fine now. Closing.

closed 2023-07-29 by fufexan