boltgolt/howdy #769: Support for `doas`

hackdev17, 2023-03-25:
With `doas` I was not able to run the /usr/bin/howdy even as a privileged user and it is same with `root`

```
[user@host ~]$ doas howdy test
doas (user@host) password: 
Please run this command as root:

        sudo howdy test
[user@host ~]$ su -
Password: 
[root@host ~]# howdy test
Please run this command as root:

        sudo howdy test
``` 
Suggest any configuration changes if required

boltgolt, 2023-03-25:
That's weird because howdy checks if .geteuid() returns 0. What is your root user id?

hackdev17, 2023-03-25:
As a normal user my id is 1000

hackdev17, 2023-03-26:
I have tested `howdy` with `sudo` it works fine though, any file that invoke `sudo` I havent found yet, where do the `sudo` get invoked in the process `sudo howdy test`

boltgolt, 2023-03-26:
Sudo doesn't get invoked, in a simplified way: By putting `sudo` before a command that command is executed as if the root user has executed it, so the user id of the process is 0

What does `doas id` give you?

hackdev17, 2023-03-26:
```
[user@host ~]$ doas id
doas (user@host) password: 
uid=0(root) gid=0(root) groups=0(root)
```

hackdev17, 2023-03-26:
The issue has now been resolved !
Following this guide

I have added the line to `/etc/pam.d/doas`
`auth sufficient pam_python.so /lib/security/howdy/pam.py`

```
[user@host ~]$ doas chmod -R 755 /lib/security/howdy
[user@host ~]$ cat /etc/pam.d/doas 
#%PAM-1.0
auth sufficient pam_python.so /lib/security/howdy/pam.py
auth            include         system-auth
account         include         system-auth
session         include         system-auth
```

```
[user@host ~]$ doas ls /
Attempting face detection
[ WARN:0@0.048] global cap_gstreamer.cpp:2784 handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.048] global cap_gstreamer.cpp:1679 open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.048] global cap_gstreamer.cpp:1164 isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
Identified face as user
bin  boot  dev  etc  home  lib  lib64  lost+found  mnt  opt  proc  root  run  sbin  srv  sys  tmp  usr  var
```

However I have to configure faces using `sudo` I can't configure using `doas` for now
with `doas` I was only able to edit the config files

hackdev17, 2023-03-26:
Also there is no `geteuid()` in the `/usr/bin/howdy`, that was the reason I think it was not working with `doas`

```
[user@host ~]$ sudo howdy version
Howdy 2.6.1
[user@host ~]$ cat /usr/bin/howdy | grep geteuid | wc -l
0
```

musikid, 2023-10-13:
Closed by #798.

closed 2023-10-13 by musikid