boltgolt/howdy #771: command not found: howdy

colawithsauce, 2023-03-29:
My distribution: ubuntu 22.02.2 LTS

I install it by following the introduction of README.md: adds ppa source and install it. (but when installing the three data pack, I download then manually instead.) 

I have tried to list files of howdy, and also didn't find a excutable file
```bash
➜  ~ dpkg  -L howdy
/.
/lib
/lib/security
/lib/security/howdy
/lib/security/howdy/recorders
/lib/security/howdy/recorders/ffmpeg_reader.py
/lib/security/howdy/recorders/pyv4l2_reader.py
/lib/security/howdy/recorders/v4l2.py
/lib/security/howdy/recorders/video_capture.py
/lib/security/howdy/recorders/__init__.py
/lib/security/howdy/pam.py
/lib/security/howdy/logo.png
/lib/security/howdy/cli.py
/lib/security/howdy/dlib-data
/lib/security/howdy/dlib-data/Readme.md
/lib/security/howdy/dlib-data/install.sh
/lib/security/howdy/compare.py
/lib/security/howdy/cli
/lib/security/howdy/cli/add.py
/lib/security/howdy/cli/test.py
/lib/security/howdy/cli/list.py
/lib/security/howdy/cli/disable.py
/lib/security/howdy/cli/__init__.py
/lib/security/howdy/cli/config.py
/lib/security/howdy/cli/snap.py
/lib/security/howdy/cli/clear.py
/lib/security/howdy/cli/remove.py
/lib/security/howdy/snapshot.py
/lib/security/howdy/pam-config
/lib/security/howdy/pam-config/howdy
/lib/security/howdy/config.ini
/usr
/usr/share
/usr/share/doc
/usr/share/doc/howdy
/usr/share/doc/howdy/changelog.gz
/usr/share/doc/howdy/copyright
/usr/share/man
/usr/share/man/man1
/usr/share/man/man1/howdy.1.gz
/usr/share/bash-completion
/usr/share/bash-completion/completions
/usr/share/bash-completion/completions/howdy
/usr/share/pam-configs
/usr/share/pam-configs/howdy
/usr/share/lintian
/usr/share/lintian/overrides
/usr/share/lintian/overrides/howdy
```

So what the problem with my installation?

Ozy-Viking, 2023-03-31:
The executable `howdy` is a link to the python file located in `/lib/security/howdy/cli.py`

```bash
$ where howdy
/usr/local/bin/howdy

$ ls -lah /usr/local/bin/howdy
lrwxrwxrwx  1 root root   26 Nov 13 09:40 howdy -> /lib/security/howdy/cli.py*
```

Try running the `ln` command to create that link.
```bash
sudo ln /lib/security/howdy/cli.py /usr/local/bin/howdy
```

etx2, 2023-04-08:
I have also same problem, just create a symlink, your suggestion @Ozy-Viking is the way but for me do not work without use the -s option 
`sudo ln -s /lib/security/howdy/cli.py /usr/local/bin/howdy`

Thanks for the tips!

Shaddy-ElDesouky, 2024-07-09:
I have the same problem, I found /lib/security/howdy/cli.py but could not find /usr/local/bin/howdy

 or /usr/bin/howdy , I started my problem in #923

closed 2026-11-16 by boltgolt