boltgolt/howdy #774: howdy fails to install with pip externally-managed-environment

rayzorben, 2023-04-03:
Setting up howdy (2.6.1) ...

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installatio
n or OS distribution provider. You can override this, at the risk of breaking 
your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error ex
it status 1
Processing triggers for man-db (2.11.2-1) ...
Processing triggers for libc-bin (2.37-0ubuntu2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

wangyuhuiever, 2023-04-07:
No LSB modules are available.
Distributor ID:	Ubuntu
Description:	Ubuntu Lunar Lobster (development branch)
Release:	23.04
Codename:	lunar

etx2, 2023-04-08:
You can disable the externally-managed-environment (it is improved in the latest version of python). 
rename -this file /usr/lib/python3.x/EXTERNALLY-MANAGED and after install howdy.

Jackaed, 2023-04-19:
Can you please elaborate on this? I don't understand what you mean or what this achieves.

perlun, 2023-04-20:
I am also seeing the above on Debian bookworm. Temporarily moving away `/usr/lib/python3.11/EXTERNALLY-MANAGED` helped "fix" the problem (but bear in mind that this clearly feels like the "wrong" way to fix it when `pip` is actually provided by the system). 

(I got dlib-related errors after "fixing" this, #726)

etx2, 2023-04-20:
> I am also seeing the above on Debian bookworm. Temporarily moving away `/usr/lib/python3.11/EXTERNALLY-MANAGED` helped "fix" the problem (but bear in mind that this clearly feels like the "wrong" [...]

Yes, I know it is the wrong way to workaround the problem but I don t understand how can we use python in "virtual ambient" with howdy

CodeOn-ArK, 2023-04-21:
Any good fix for this?

raknerak, 2023-04-24:
I have the same issue after upgrading Ubuntu 22.10 to 23.04. It would be great to have permanent solution.

fenollp, 2024-07-17:
Same installation error on `24.04` Noble:
```
# sudo apt install howdy
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Suggested packages:
  nvidia-cuda-dev
The following NEW packages will be installed
  howdy
0 to upgrade, 1 to newly install, 0 to remove and 0 not to upgrade.
Need to get 0 B/38.9 kB of archives.
After this operation, 166 kB of additional disk space will be used.
Selecting previously unselected package howdy.
(Reading database ... 295958 files and directories currently installed.)
Preparing to unpack .../archives/howdy_2.6.1_all.deb ...
Starting IR camera check...

Trying "Integrated Camera: Integrated C"
One of your cameras should now be on.
Did your IR emitters turn on? [y/N]: y

Starting certainty auto config...

After detection, Howdy knows how certain it is that the match is correct.
How certain Howdy needs to be before authenticating you can be customized.

F: Fast.
Allows more fuzzy matches, but speeds up the scanning process greatly.

B: Balanced.
Still relatively quick detection, but might not log you in when further away.

S: Secure.
The safest option, but will take much longer to authenticate you.

You can always change this setting in the config.
What profile would you like to use? [f/b/s]: f

Unpacking howdy (2.6.1) ...
Setting up howdy (2.6.1) ...
/var/lib/dpkg/info/howdy.postinst:145: SyntaxWarning: invalid escape sequence '\w'
  "davisking-dlib-\w+/(dlib/(http_client|java|matlab|test/)|"

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.12/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override 
this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.12.0-4build2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

boltgolt, 2026-11-16:
Hey, thanks for reporting this! This was reported against Howdy v2. Because v3 introduces a new PAM integration and installation, I think this issue is now stale. Please try the newest version and open a new issue if it's still relevant there!

closed 2026-11-16 by boltgolt

lukaszsobala, 2026-11-16:
Not sure if related, but maybe?
I have encountered this error on Ubuntu 26.04:
```bash
>sudo apt install howdy-gtk

Solving dependencies... Error!  
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

Unsatisfied dependencies:
 howdy-gtk : Depends: howdy but it is not installable
             Depends: python3-pip but it is not going to be installed
Error: Unable to satisfy dependencies. Reached two conflicting assignments:
   1. howdy-gtk:amd64=1.0.0 is selected for install
   2. howdy-gtk:amd64 Depends howdy
      but none of the choices are installable:
      [no choices]
```
It may be because I have `amd64v3` packages enabled. `python3-pip` installs only as `i386`. But maybe it's not because of this.