boltgolt/howdy #776: `sudo howdy test` giving error

hemangjoshi37a, 2023-04-05:
Comamnd I ran : 
```bash
sudo howdy test
```

Error I got : 
```python
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 109, in <module>
    import cli.test
  File "/usr/lib/security/howdy/cli/test.py", line 8, in <module>
    import dlib
ModuleNotFoundError: No module named 'dlib'
```

hemangjoshi37a, 2023-04-05:
Comamnd I ran : 
```bash
sudo howdy -U hemang add
```

Error I got : 
```python
No module named 'dlib'

Can't import the dlib module, check the output of
pip3 show dlib
```

etx2, 2023-04-10:
you must install dlib from pip
`pip install dlib`

etx2, 2023-04-10:
what the return of 
`pip show dlib`
if u have newer version of debian you can't install whitout disable the externally-managed-environment (this is marked as default for security)
please show me 
`ls -l /usr/local/bin/howdy `

Jackaed, 2023-04-19:
I'm having this same issue. I believe it occurs at installation, as a number of errors are spat out related to dlib.

Here's the output from the install:

https://pastebin.com/V87Vgcuu

The output from pip show dlib:

`Name: dlib
Version: 19.24.1
Summary: A toolkit for making real world machine learning and data analysis applications
Home-page: https://github.com/davisking/dlib
Author: Davis King
Author-email: davis@dlib.net
License: Boost Software License
Location: /home/edward/.local/lib/python3.11/site-packages
Requires:
Required-by:`

The output from `ls -l /usr/local/bin/howdy`:

`lrwxrwxrwx 1 root root 26 Apr 19 16:38 /usr/local/bin/howdy -> /lib/security/howdy/cli.py*`

Jackaed, 2023-04-19:
Also noteworthy is that when uninstalling howdy you get:

```WARNING: Skipping dlib as it is not installed.```

etx2, 2023-04-20:
try to remove it and install manually via pip install dlib. 
I have also installed this python3-full in my sistem from apt. 
My running version is different from yours and I have 19.16.0

closed 2026-11-16 by boltgolt