boltgolt/howdy #778: Use with external face recognition system

kostyawhite, 2023-04-15:
I just wonder how hard would it be to make HOWDY work with external face recognition system, via api for example. So basically with need to implement several methods like add-face, create-vector, search-face, any thoughts on this? Thanks

boltgolt, 2023-04-15:
What would be the benefit of this?

kostyawhite, 2023-04-16:
Hi, so in some organization we have central server for access control, with all employees in database, we can remotely give or revoke permissions to access certain areas of the building and so on, and in this case we would be able to revoke access to working pc's. Face recognition system would be running locally as well. 

Currently I see only one option to do so, it is to rewrite compare.py