boltgolt/howdy #781: Problems running on Ubuntu 23.04 (lunar)

tante, 2023-04-20:
Ubuntu 23.04 comes with a new version of Python (default is 3.11) that makes howdy not run properly.

The first issue is line 10 in pam.py:
`import ConfigParser`
The module is now called `configparser` so howdy can't properly import it.

Doing a mitigation by changing that line to `import configparser as ConfigParser` makes howdy work in GDM for login but when using it in sudo creates a funky error:

`Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 23, in <module>
    from recorders.video_capture import VideoCapture`

The strange thing is: If you add `import recorders` to the file there is no problem. Just importing that submodule fails and seemingly only while trying it via sudo. 

----

Linux distribution (if applicable): Ubuntu 23.04

Howdy version (`sudo howdy version`): Howdy 2.6.1

ashkop, 2023-04-21:
After changing the `configparser` import I had a different issue:
```
ModuleNotFoundError: No module named 'ConfigParser'
```
After installing dlib with `sudo pip install --break-system-packages dlib` everything is working now.

kimhongsu, 2023-04-21:
@ashkop You made my day. 
Howdy is working fine in my Kubuntu 23.04 now. 
```
sudo apt install -y python3-opencv
sudo pip install --break-system-packages dlib
sudo howdy add
```

ashkop, 2023-04-21:
@kimhongsu I'm glad that helped. Would still be nice if howdy can be fully updated to python 3.

adamlove86, 2023-04-22:
Purge and reinstall seems to have run into errors:

```
Setting up howdy (2.6.1) ...

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.11.2-1) ...
Processing triggers for libc-bin (2.37-0ubuntu2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

```

mstrobl2, 2023-04-23:
Here's my ugly hack workaround that also works for fresh install of howdy on 23.04, just skip the "apt remove howdy" command.

`sudo -i`
`mv /usr/lib/python3.11/EXTERNALLY-MANAGED /usr/lib/python3.11/DISABLED-EXTERNALLY-MANAGED`
`apt remove howdy`
`pip3 install dlib`
`apt install howdy`
`nano /lib/security/howdy/pam.py # Change ConfigParser to configparser on 2 lines` 
`mv /usr/lib/python3.11/DISABLED-EXTERNALLY-MANAGED /usr/lib/python3.11/EXTERNALLY-MANAGED`

skjerns, 2023-04-23:
> nano /lib/security/howdy/pam.py # Change ConfigParser to configparser on 2 lines

note: You also need to change it elsewhere, or just change the line to `import configparser as ConfigParser`

boltgolt, 2023-04-23:
This is an annoying one, using `--break-system-packages` should help but it's really not the clean way to do it. Maybe a venv for Howdy would be better

Arad-Afzali, 2023-04-27:
I've recently updated to Ubuntu 23.04 and Howdy is not working on the lock screen. I tried to reinstall it but it didn't work. 
I can still add new faces and Howdy seems to work fine in the terminal and gives me no errors.

dingwen07, 2023-05-18:
`--break-system-packages` is not a good practice. Maybe howdy could use virtual environments to install this package.

nate3D, 2023-05-26:
> Purge and reinstall seems to have run into errors:
>
> ```
> Setting up howdy (2.6.1) ...
>
>
> error: externally-managed-environment
>
> × This environment is externally managed
> ╰─> To install Python packages [...]

I've got the exact same issue with a fresh install of Ubuntu 23.04 Linux 6.2.0-20-generic 64-bit. Wrapping howdy inside of it's own venv would be the optimal solution here and for isolating its install further from the host.

MurzNN, 2023-06-10:
@nate3D  Thanks for the solution, could you please share an example of how to do the "Wrapping howdy inside of it's own venv"?

MurzNN, 2023-06-11:
I've found a solution to the `ModuleNotFoundError: No module named 'ConfigParser'` error, the fix is in the https://github.com/boltgolt/howdy/issues/799

Rongronggg9, 2023-06-21:
To those who are not willing to use `--break-system-packages`:

```sh
sudo pip install -t /lib/security/howdy dlib
```

Tested on the `beta` branch, should work on `master`.

kimhongsu, 2023-08-19:
> This is an annoying one, using `--break-system-packages` should help but it's really not the clean way to do it. Maybe a venv for Howdy would be better

@boltgolt Could you let us know how to wrap howdy up with venv? Or is it possible to wrap howdy up with docker?

Spurlos, 2023-09-22:
> To those who are not willing to use `--break-system-packages`:
> 
> ```shell
> sudo pip install -t /lib/security/howdy dlib
> ```
> 
> Tested on the `beta` branch, should work on `master`.

@boltgolt Perhaps this fix can be incorporated in release or at least beta?

g1ronn1mo, 2023-12-10:
> > To those who are not willing to use `--break-system-packages`:
> > ```shell
> > sudo pip install -t /lib/security/howdy dlib
> > ```
> > 
> > 
> >     
> >       
> >     
> > 
> >       
> >     
> > 
> >     
> >   
> > Tested on the `beta` branch, should work on `master`.
> 
> @boltgolt Perhaps this fix can be incorporated in release or at least beta?

Tried on ubuntu 23.10 Could install howdy after installing dlib like this, but could not run howdy.

4mritGiri, 2023-12-11:
> Purge and reinstall seems to have run into errors:
>
> ```
> Setting up howdy (2.6.1) ...
>
>
> error: externally-managed-environment
>
> × This environment is externally managed
> ╰─> To install Python packages [...]

I am also get the same problem on Ubuntu version `23.10.1`  

## Howdy Installation Issue on Ubuntu 23.10.1

### Problem:
I encountered an issue with Howdy installation on Ubuntu version `23.10.1`. The provided solutions seemed to have complications.

### Workaround:
I found a workaround that worked for me, even on a fresh install of Ubuntu 23.10.1. Skip the "apt remove howdy" command to avoid additional errors during sudo authentication.

1. Open a root shell:
   ```sh
   sudo -i

2. Move the EXTERNALLY-MANAGED directory:
   ```sh 
     mv /usr/lib/python3.11/EXTERNALLY-MANAGED /usr/lib/python3.11/DISABLED-EXTERNALLY-MANAGED
  
3. Remove Howdy (skip if fresh install):
   ```sh
    apt remove howdy

4. Install dlib using pip3:
     `pip3 install dlib`

   If pip3 doesn't work, try `pip install dlib`. If that also fails, attempt:
      `sudo pip install --break-system-packages dlib`

5. Reinstall Howdy:
   ```sh
    apt install howdy

8. Edit pam.py to fix import error:
   ```sh
    nano /lib/security/howdy/pam.py

 Change `import ConfigParser` to `import configparser as ConfigParser` you can find on lines 2 to 11.

9. Restore EXTERNALLY-MANAGED directory:
   ```sh
    mv /usr/lib/python3.11/DISABLED-EXTERNALLY-MANAGED /usr/lib/python3.11/EXTERNALLY-MANAGED 

This workaround should help resolve the installation issues in my case. 

While adding face some time you will get device_path problem but i am solve it like this.
   ```sh
    # Should be set automatically by an installer if your distro has one
    # device_path = /dev/v4l/by-path/pci-0000:00:14.0-usb-0:6:1.2-video-index1 # default in my config comment it add new below like this
    device_path = /dev/video0

g1ronn1mo, 2023-12-11:
https://github.com/boltgolt/howdy/issues/851

I also have a silution without breaking system packages. This solution depends on a dep package which is not released yet.

closed 2026-11-16 by boltgolt