boltgolt/howdy #782: Add passkey webauthn and the option to use hardware keys (example yubikeys) in order to make the login process more secure.

Chummpp, 2023-04-23:
Howdy seems to be the only facial recognition technology for mainstream Linux use and it can easily be one of the best Linux security tools. Howdy could make Linux more secure by making use of passkey webauthn. Also as a side note don't hold your breath for Ubuntu to implement facial recognition with passkey or any other form of passkey for that matter. Ubuntu and all the other Linux OS/distro developers are either lazy when it comes to implementing security features for mainstream use or it doesn't really care about implementing security features when it comes to creating a operating system for the average user at home.

boltgolt, 2023-04-23:
Howdy is just a PAM module for face authentication. If you want to use another type (like fingerprint, or hardware keys) you need another PAM module for that. Most of what you're talking about already exists but is not installed by default

closed 2023-04-23 by boltgolt

Chummpp, 2023-04-23:
> Most of what you're talking about already exists but is not installed by default

If that's true than why is it not widely known, talked about, and not already pre installed for the user?

boltgolt, 2023-04-23:
Can't say, i don't maintain any distros