boltgolt/howdy #784: Fedora 38 - detected unhandled Python exception in '/lib64/security/howdy/compare.py'

elegos, 2023-04-24:
Hello!

I've just installed Howdy on a Fedora 38 installation via the copr repository. I've setup pam.d/sudo to use howdy and, after manually setting the right /dev/videoX device in the configuration, the webcam effectively activates when a sudo command is being prompted. The problem is that it will fail and fallback to the password method.

journalctl reports me the following:

```apr 24 14:48:58 plungedcopper [14430]: Attempting facial authentication for user elegos
apr 24 14:49:01 plungedcopper python3[14432]: detected unhandled Python exception in '/lib64/security/howdy/compare.py'
apr 24 14:49:01 plungedcopper [14430]: Failure, unknown error1
apr 24 14:49:01 plungedcopper abrt-server[14498]: Package 'howdy' isn't signed with proper key
apr 24 14:49:01 plungedcopper abrt-server[14498]: 'post-create' on '/var/spool/abrt/Python3-2023-04-24-14:49:01-14432' exited with 1
apr 24 14:49:01 plungedcopper abrt-server[14498]: Deleting problem directory '/var/spool/abrt/Python3-2023-04-24-14:49:01-14432'```

I don't know how to view that unhandled exception honestly (Fedora doesn't have `/var/log/auth.log` or similar).

May you please help me?
Thanks :)

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Fedora 38

Howdy version (`sudo howdy version`): 2.6.1

elegos, 2023-04-24:
Ha! Found the error. I tried to run `python3 compare.py elegos` and actually I saw what resulted being the problem:

```libva info: VA-API version 1.18.0
libva info: Trying to open /usr/lib64/dri/nvidia_drv_video.so
libva info: va_openDriver() returns -1
MFX: Can't initialize session
```

The issue is that apparently I had to install the `nvidia-vaapi-driver` package.

jimpudar, 2023-04-28:
I also had the same symptom on Fedora 38, but my exception was a little different:

```
Traceback (most recent call last):
  File "/lib64/security/howdy/compare.py", line 283, in <module>
    make_snapshot("SUCCESSFUL")
  File "/lib64/security/howdy/compare.py", line 55, in make_snapshot
    snapshot.generate(snapframes, [
  File "/usr/lib64/security/howdy/snapshot.py", line 54, in generate
    os.makedirs(abpath + "/snapshots")
  File "<frozen os>", line 225, in makedirs
PermissionError: [Errno 13] Permission denied: '/usr/lib64/security/howdy/snapshots'
```

This was resolved by running `sudo howdy snapshot`, which created the non-existing directory. Everything worked properly after that!

xuanvinhvu, 2023-05-04:
> I also had the same symptom on Fedora 38, but my exception was a little different:
>
> ```
> Traceback (most recent call last):
>   File "/lib64/security/howdy/compare.py", line 283, in <module> [...]

Nice! I had the same problem and running the snapshot resolved it.

jorgeperezlara, 2023-05-17:
> I also had the same symptom on Fedora 38, but my exception was a little different:
>
> ```
> Traceback (most recent call last):
>   File "/lib64/security/howdy/compare.py", line 283, in <module> [...]

Thanks!!! This solved it for me!

stephanlamoureux, 2023-05-31:
> I also had the same symptom on Fedora 38, but my exception was a little different:
>
> ```
> Traceback (most recent call last):
>   File "/lib64/security/howdy/compare.py", line 283, in <module> [...]

Thanks! This and the ConfigHowdy bash script did the trick.

closed 2026-11-14 by boltgolt