boltgolt/howdy #786: Howdy dlib error - no module

Hamster1962, 2023-05-05:
My Howdy was working perfectly until a large update last night. Lots of Python and Camera type updates.
Now when i try and use Howdy i get the error :-

Traceback (most recent call last):
File "/lib/security/howdy/compare.py", line 18, in <module>
import dlib
ModuleNotFoundError: No module named 'dlib'

Restoring a Snapshot solves the issue until the next update is run
----

I've searched for similar issues already, and my issue has not been reported yet. Yes

Linux distribution = Garuda Linux based on Arch with KDE Desktop Kernel 6.2.13 zen-1-zen

Howdy version (`sudo howdy version`): Howdy 3.0.0 BETA

Please help

renyuneyun, 2023-05-06:
Did you rebuild the relevant python packages after system update?

In my case (also yesterday with arch updates), I have to rebuild some relevant packages, though not dlib.

letavocado, 2023-05-14:
@Hamster1962 try `sudo pip install dlib`

letavocado, 2023-05-14:
Magic!  :smile: :magic_wand:

PabloRuizCuevas, 2023-07-06:
this worked: but in a bit more aggressive version for new versions of ubuntu/pip/python:

`sudo pip install dlib --break-system-packages`

please take into account that this may break existing installation of the system, so it is maybe a bit risky for your system

jackfido, 2023-11-29:
So, in my case, after check my error I found that C compiler was not found, I had to install build essentials with `sudo nala install build-essential` (or it can be by `sudo apt install build-essential`)

Here was my error at install dlib:

```
goliapp@goliapp-lenovothinkbook:~$ sudo pip install dlib --break-system-packages
Collecting dlib
  Using cached dlib-19.24.2.tar.gz (11.8 MB)
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: dlib
  Building wheel for dlib (pyproject.toml) ... error
  error: subprocess-exited-with-error
  
  × Building wheel for dlib (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [68 lines of output]
      running bdist_wheel
      running build
      running build_ext
      Building extension for Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0]
      Invoking CMake setup: 'cmake /tmp/pip-install-40ergljx/dlib_7d8202288724454bb15589a4b9b555a2/tools/python -DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-install-40ergljx/dlib_7d8202288724454bb15589a4b9b555a2/build/lib.linux-x86_64-cpython-310 -DPYTHON_EXECUTABLE=/usr/bin/python3 -DCMAKE_BUILD_TYPE=Release'
      -- The C compiler identification is GNU 11.4.0
      **-- The CXX compiler identification is unknown**
      -- Detecting C compiler ABI info
      -- Detecting C compiler ABI info - done
      -- Check for working C compiler: /usr/bin/cc - skipped
      -- Detecting C compile features
      -- Detecting C compile features - done
      CMake Error at CMakeLists.txt:14 (project):
        No CMAKE_CXX_COMPILER could be found.
      
        Tell CMake where to find the compiler by setting either the environment
        variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
        to the compiler, or to the compiler name if it is in the PATH.
      
      
      -- Configuring incomplete, errors occurred!
      Traceback (most recent call last):
        File "/usr/local/lib/python3.10/dist-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 353, in <module>
          main()
        File "/usr/local/lib/python3.10/dist-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 335, in main
          json_out['return_val'] = hook(**hook_input['kwargs'])
        File "/usr/local/lib/python3.10/dist-packages/pip/_vendor/pyproject_hooks/_in_process/_in_process.py", line 251, in build_wheel
          return _build_backend().build_wheel(wheel_directory, config_settings,
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/build_meta.py", line 404, in build_wheel
          return self._build_with_temp_dir(
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/build_meta.py", line 389, in _build_with_temp_dir
          self.run_setup()
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/build_meta.py", line 311, in run_setup
          exec(code, locals())
        File "<string>", line 218, in <module>
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/__init__.py", line 103, in setup
          return distutils.core.setup(**attrs)
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/_distutils/core.py", line 185, in setup
          return run_commands(dist)
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/_distutils/core.py", line 201, in run_commands
          dist.run_commands()
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py", line 969, in run_commands
          self.run_command(cmd)
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/dist.py", line 963, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/wheel/bdist_wheel.py", line 368, in run
          self.run_command("build")
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/dist.py", line 963, in run_command
          super().run_command(command)
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/_distutils/dist.py", line 988, in run_command
          cmd_obj.run()
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/_distutils/command/build.py", line 131, in run
          self.run_command(cmd_name)
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/_distutils/cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "/tmp/pip-build-env-5s_8h49g/overlay/local/lib/python3.10/dist-packages/setuptools/dist.py", line 963, in run_command
          super().run_command(command)
[... cut ...]
      subprocess.CalledProcessError: Command '['cmake', '/tmp/pip-install-40ergljx/dlib_7d8202288724454bb15589a4b9b555a2/tools/python', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=/tmp/pip-install-40ergljx/dlib_7d8202288724454bb15589a4b9b555a2/build/lib.linux-x86_64-cpython-310', '-DPYTHON_EXECUTABLE=/usr/bin/python3', '-DCMAKE_BUILD_TYPE=Release']' returned non-zero exit status 1.
      [end of output]
  
  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for dlib
Failed to build dlib
ERROR: Could not build wheels for dlib, which is required to install pyproject.toml-based projects
```

After run build essential all run ok

boltgolt, 2026-11-16:
Hey, thanks for reporting this! This was reported against an install method from before v3. Because v3 introduces a new PAM integration and installation, I think this issue is now stale. Please try the newest version and open a new issue if it's still relevant there!

closed 2026-11-16 by boltgolt