boltgolt/howdy #790: Camera not working on Lenovo X1 Carbon 6th Gen on Fedora 38

ekx, 2023-05-12:
I've installed howdy via copr on Fedora 38 on my Lenovo X1 Carbon 6th Gen. When I try to add a face model the IR emitters turn on and after a couple of seconds I get the following error:

```
[ben@laptop ~]$ sudo howdy add
Adding face model for the user ben
Enter a label for this new model [Initial model] (max 24 characters): 
[ INFO:0@2.445] global videoio_registry.cpp:232 VideoBackendRegistry VIDEOIO: Enabled backends(9, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); V4L2(970); CV_IMAGES(960); CV_MJPEG(950); FIREWIRE(940); UEYE(930); OBSENSOR(920)
[ INFO:0@2.445] global backend_plugin.cpp:369 getPluginCandidates VideoIO plugin (FFMPEG): glob is 'libopencv_videoio_ffmpeg*.so', 1 location(s)
[ INFO:0@2.458] global backend_plugin.cpp:379 getPluginCandidates     - /lib64: 0
[ INFO:0@2.458] global backend_plugin.cpp:383 getPluginCandidates Found 0 plugin(s) for FFMPEG
[ INFO:0@2.471] global cap_gstreamer.cpp:1403 open OpenCV | GStreamer: /dev/video0
[ INFO:0@2.471] global cap_gstreamer.cpp:1436 open OpenCV | GStreamer: mode - FILE
[ WARN:0@2.573] global cap_gstreamer.cpp:2784 handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@2.574] global cap_gstreamer.cpp:1679 open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@2.574] global cap_gstreamer.cpp:1164 isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
libva info: VA-API version 1.18.0
libva info: Trying to open /usr/lib64/dri/iHD_drv_video.so
libva info: Found init function __vaDriverInit_1_18
libva info: va_openDriver() returns 0
MFX: Unsupported extension: /dev/video0
[ WARN:0@12.752] global cap_v4l.cpp:1119 tryIoctl VIDEOIO(V4L2:/dev/video0): select() timeout.

Please look straight into the camera
[ WARN:0@24.764] global cap_v4l.cpp:1119 tryIoctl VIDEOIO(V4L2:/dev/video0): select() timeout.
Failed to read camera specified in your 'device_path', aborting
```
The webcam only supports one resolution
```
[ben@laptop ~]$ v4l2-ctl -d /dev/video0 --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'YUYV' (YUYV 4:2:2)
		Size: Discrete 160x120
			Interval: Discrete 0.033s (30.000 fps)
```
I have tried turning setting this resolution in howdy's config and turning on ```force_mjpeg``` both to no avail.
I also tried setting the resolution to 400 by 400 as suggested by this [wiki article](https://wiki.archlinux.org/title/Lenovo_ThinkPad_X1_Carbon_(Gen_6)) but that didn't change anything either.

If the wiki article is to be believed the hardware worked at one point with howdy... at least on arch linux.
So maybe this is a bug/regression?

----

I've searched for similar issues already, and my issue has not been reported yet: Yes

Linux distribution (if applicable): Fedora 38

Howdy version (`sudo howdy version`): 2.6.1

tokox, 2023-08-16:
>  Failed to read camera specified in your 'device_path', aborting

Are you sure camera device in device_path is correct? Does reading from that device from other programs (eg. video player) work?

boredland, 2023-11-26:
Not on fedora, but on arch, this configuration works fine. 

1. Make sure you have all the latest firmware updates, as in my experience that makes a big difference with a lot of lenovo devices!
2. Make sure you have those configurations: 
   ```
   device_path = /dev/v4l/by-id/usb-SunplusIT_Inc_Integrated_IR_Camera-video-index0
   max_height = 400
   force_mjpeg = true
   ```

ekx, 2023-11-30:
@tokox pretty sure the path I was using was correct yes. When I try to use the device from other programs the red LEDs next to the camera turn on but I don't get a video feed... I don't know if this is to be expected for IR cameras. I set the resolution already.

@boredland This is what 'sudo fwupdmgr update' says: 
`Devices with no available firmware updates: 
 • INTEL SSDPEKKF256G8L
Devices with the latest available firmware version:
 • Embedded Controller
 • Intel Management Engine
 • System Firmware
 • UEFI dbx
`
So I guess I am at the latest firmware? Also I tried with your exact config and now I get the following error when I run "sudo howdy add" or "sudo howdy test"
`Camera path is not configured correctly, please edit the 'device_path' config value.
Exception ignored in: <function VideoCapture.__del__ at 0x7f11c728d760>
Traceback (most recent call last):
  File "/usr/lib64/security/howdy/recorders/video_capture.py", line 55, in __del__
    self.internal.release()
    ^^^^^^^^^^^^^
AttributeError: 'VideoCapture' object has no attribute 'internal'
`

boredland, 2023-11-30:
Well, I am on arch - perhaps they behave differently. What does `/dev/v4l/by-id/` yield?

ekx, 2023-11-30:
```
[ben@laptop ~]$ ls /dev/v4l/by-id
usb-Chicony_Electronics_Co._Ltd._Integrated_Camera-video-index0  usb-SunplusIT_Inc_SPCA2085_PC_Camera_01.00.00-video-index0
usb-Chicony_Electronics_Co._Ltd._Integrated_Camera-video-index1  usb-SunplusIT_Inc_SPCA2085_PC_Camera_01.00.00-video-index1
```

closed 2026-11-16 by boltgolt