boltgolt/howdy #791: Camera not turning on on the lockscreen (kali 6.1.0)

vyPal, 2023-05-13:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

Hello, I have just installed a fresh install of kali. Howdy is the first program that I have installed. I ran into some issues when installing, but I was ale to fix them with help from existing issues. I have successfully added 2 faces, and used `sudo howdy test` and tested that everything works correctly. When using `sudo howdy test` the camera indicator turns on, the IR emitter start flashing, and I can see my face perfectly in the test window. It does detect both faces too. The issue comes when locking the screen. After I lock the screen and get to the login page (by locking the screen I mean both pressing the lock button in the UI and shutting the laptop) I get presented with the standard login screen, however the camera indicator doesn't turn on, nor does the IR emitter start flashing. I have tried setting `no_confirmation` to true, Running  `sudo chmod -R 755 /lib/security/howdy` and some other things, including reinstalling howdy. Also `/var/log/auth.log` does not exist. (The folder exists and has files in it, but there is no auth.log) I have not been able to locate any errors, but I also don't really know where to look for them.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Kali 2023.1

Howdy version (`sudo howdy version`): Howdy 2.6.1

eightfy, 2023-05-25:
I had a similar issue and resolved it after reinstalling pam-python. Hope this helps you.

techonerd, 2023-11-20:
I was facing the same issue, using `sddm` on kali.
This worked for me [kde-plasma-fix](https://github.com/boltgolt/howdy/wiki/Howdy-on-Debian-Testing---KDE-Plasma#workaround-for-kde-plasma)

Hope it helps. :)

closed 2023-11-20 by vyPal