boltgolt/howdy #792: unable to make pam use howdy

arthur19q3, 2023-05-15:
i installed howdy and passed the howdy test successfully and then i added the following line to the /etc/pam.d/sudo file
`auth 		sufficient 	pam_python.so	/lib/security/howdy/pam.py
`
but sudo still then asks passwd all the time.

anyone has a clue?

closed 2023-05-15 by arthur19q3

gautierlabarre, 2023-12-12:
Hello, did you figure it out ? I think I have the same issue

arthur19q3, 2023-12-14:
> Hello, did you figure it out ? I think I have the same issue

probably check your Python version and if it is compatible with what's written there in the python script. sorry mate it's been a while and my memory's a bit vague.

gautierlabarre, 2023-12-14:
It's ok, I figured it out, I had to modify some python file. It was another issue, don't remember which. Thanks :)

arthur19q3, 2023-12-14:
care to share what did the trick?  we occasionally need to hop between distros in our company and Howdy would intetestingly trigger different errors when it doesn't work out of box.I recall it's usually something to do with the version though.

gautierlabarre, 2023-12-15:
Sure the fix for me was this one : https://github.com/boltgolt/howdy/issues/785#issuecomment-1533002545

arthur19q3, 2023-12-15:
> Sure the fix for me was this one : [#785 (comment)](https://github.com/boltgolt/howdy/issues/785#issuecomment-1533002545)

thanks mate, i have encountered this some time ago as well, fix was the same.