boltgolt/howdy #799: Error on running with python3: "ModuleNotFoundError: No module named 'ConfigParser'" and the fix

MurzNN, 2023-06-11:
When I'm trying to log in using howdy in Ubuntu 23.04 which has python3 version by default - I've got an error:
```
2023-06-11T07:45:01.786521+04:00 Ubuntu /lib/security/howdy/pam.py[25199]: ModuleNotFoundError: No module named 'ConfigParser'
2023-06-11T07:45:01.790383+04:00 Ubuntu /lib/security/howdy/pam.py[25198]: Traceback (most recent call last):
2023-06-11T07:45:01.790469+04:00 Ubuntu /lib/security/howdy/pam.py[25198]:   File "/lib/security/howdy/pam.py", line 10, in <module>
2023-06-11T07:45:01.790496+04:00 Ubuntu /lib/security/howdy/pam.py[25198]:     import ConfigParser
2023-06-11T07:45:01.790522+04:00 Ubuntu /lib/security/howdy/pam.py[25198]: ModuleNotFoundError: No module named 'ConfigParser'
```
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 23.04

Howdy version (`sudo howdy version`): 2.6.1

MurzNN, 2023-06-11:
The 'ConfigParser' library in python3 is renamed to `configparser`, so the fix is to change the current code in file ` /usr/lib/security/howdy/pam.py` like this:
```patch
@@ -6,11 +6,17 @@ import os
 import glob
 import syslog
 
-# pam-python is running python 2, so we use the old module here
-import ConfigParser
+import sys
+# The config parser is named in python2 as ConfigParser but in python3 as configparser
+# and here is a workaround for this.
+if sys.version_info[0] == 2:
+    import ConfigParser
+    config = ConfigParser.ConfigParser()
+else:
+    import configparser
+    config = configparser.ConfigParser()
 
 # Read config from disk
-config = ConfigParser.ConfigParser()
 config.read(os.path.dirname(os.path.abspath(__file__)) + "/config.ini")
```

MurzNN, 2023-06-11:
I've found that the similar changes already implemented in the `dev` branch: https://github.com/boltgolt/howdy/blob/dev/src/pam.py#L10-L15

So, could you please make a new release with these changes?

PabloRuizCuevas, 2023-07-06:
Same exception here after updating ubuntu to 23:04, the suggested changes do solve the issue

closed 2026-11-16 by boltgolt