boltgolt/howdy #800: "Unknown error: 1" when attempting login

MrD-RC, 2023-06-14:
Please forgive me if I don't have great information here. I'm completely new to Fedora, and have only mildly dabbled with Linux in the past. This is actually the first thing I'm trying to set up.

When trying to log in via howdy. I just get a message stating **Unknown error: 1**. This applies to both sudo and the gdm login.

For installation, I followed [this](https://copr.fedorainfracloud.org/coprs/principis/howdy/) guide. I completed the main installation, and set up pam.d for sudo and gdm-password. In the first instance, I did not create the module in the SELinux section. howdy test worked fine. I could see my image on the cam and a red tracking circle around my face.

[image]

Thought there is an odd output, which may give a clue to the problem?
`[ INFO:0@36.124] global cap_v4l.cpp:1973 controlInfo VIDEOIO(V4L2:/dev/video0): property unknown is not supported`

I was able to add a model to my user. Which again, seemed to work. But had another output that makes me think it's not working properly?

```
[ INFO:0@7.610] global videoio_registry.cpp:232 VideoBackendRegistry VIDEOIO: Enabled backends(9, sorted by priority): FFMPEG(1000); GSTREAMER(990); INTEL_MFX(980); V4L2(970); CV_IMAGES(960); CV_MJPEG(950); FIREWIRE(940); UEYE(930); OBSENSOR(920)
[ INFO:0@7.610] global backend_plugin.cpp:369 getPluginCandidates VideoIO plugin (FFMPEG): glob is 'libopencv_videoio_ffmpeg*.so', 1 location(s)
[ INFO:0@7.626] global backend_plugin.cpp:379 getPluginCandidates     - /lib64: 0
[ INFO:0@7.626] global backend_plugin.cpp:383 getPluginCandidates Found 0 plugin(s) for FFMPEG
[ INFO:0@7.640] global cap_gstreamer.cpp:1403 open OpenCV | GStreamer: /dev/video0
[ INFO:0@7.640] global cap_gstreamer.cpp:1436 open OpenCV | GStreamer: mode - FILE
[ WARN:0@7.724] global cap_gstreamer.cpp:2784 handleMessage OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@7.726] global cap_gstreamer.cpp:1679 open OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@7.726] global cap_gstreamer.cpp:1164 isPipelinePlaying OpenCV | GStreamer warning: GStreamer: pipeline have not been created
libva info: VA-API version 1.18.0
libva info: Trying to open /usr/lib64/dri/nouveau_drv_video.so
libva info: Found init function __vaDriverInit_1_18
libva info: va_openDriver() returns 0
MFX: Can't initialize session

Please look straight into the camera
[ INFO:0@11.114] global registry_parallel.impl.hpp:96 ParallelBackendRegistry core(parallel): Enabled backends(2, sorted by priority): TBB(1000); OPENMP(990)
[ INFO:0@11.114] global parallel_for.tbb.hpp:54 ParallelForBackend Initializing TBB parallel backend: TBB_INTERFACE_VERSION=11103
[ INFO:0@11.114] global parallel.cpp:77 createParallelForAPI core(parallel): using backend: TBB (priority=1000)
[ INFO:0@11.121] global cap_v4l.cpp:1973 controlInfo VIDEOIO(V4L2:/dev/video0): property unknown is not supported
[ INFO:0@11.258] global cap_v4l.cpp:1973 controlInfo VIDEOIO(V4L2:/dev/video0): property unknown is not supported
[ INFO:0@11.386] global cap_v4l.cpp:1973 controlInfo VIDEOIO(V4L2:/dev/video0): property unknown is not supported
[ INFO:0@11.387] global cap_v4l.cpp:1973 controlInfo VIDEOIO(V4L2:/dev/video0): property unknown is not supported
Scan complete

Added a new model to darren
```

To try and get it working. I added the SELinux module and ran the Fedora 38 setup shell script. Neither did anything positive in resolving the issue. In fact, the error now appeared twice, due to repetition in the pam.d sudo and gdm-password files. Which I fixed.

I'm guessing the problem has something to do with OpenCV or cap_v4l? Any guidance would be most appreciated.

Linux distribution (if applicable): Fedora 38

Howdy version (`sudo howdy version`): 2.6.1

Myron2812, 2023-06-19:
I'm getting the same error too. Fedora 38, howdy version 2.6.1.

tokox, 2023-06-20:
You could check `journalctl` for `python3: detected unhandled Python exception in '/lib64/security/howdy/compare.py'`. If you have this error, then #801

MrD-RC, 2023-06-20:
Thank you @tokox 

That was the solution for me too.

closed 2023-06-20 by MrD-RC