boltgolt/howdy #801: detected unhandled Python exception in '/lib64/security/howdy/compare.py'

aszabo70, 2023-06-15:
Hi,

I installed and set up howdy in Fedora 38, (Thinkad X280), but I get the following error in journalctl:

jún 15 20:10:14 thinkpad [11777]: Failure, unknown error1
jún 15 20:10:14 thinkpad python3[11779]: detected unhandled Python exception in '/lib64/security/howdy/compare.py'
jún 15 20:10:09 thinkpad [11777]: Attempting facial authentication for user xxx

Howdy version is 2.6.1-8.

No more info, what is the solution?

Thanks in advance.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version (`sudo howdy version`):

jcst-sdir, 2023-06-17:
Exactly same error for me. I am on a Thinkpad P16s. Same Howdy version. Kernel 6.3.8-200

tokox, 2023-06-20:
Same for me. Fedora 37, Howdy 2.6.1

tokox, 2023-06-20:
OK, looks like I have solution.
Howdy is trying to save snapshot to `/usr/lib64/security/howdy/snapshots/` but it doesn't exist, so it tryies to create directory, but fails for `Permission denied`. You could just create it manually and it will work.
Try `sudo mkdir /usr/lib64/security/howdy/snapshots`. Works for me.
Of course there should be solution to not have to manually create this directory, but it also works.

MrD-RC, 2023-06-20:
This was the solution for me too. Thank you!!

boltgolt, 2023-06-24:
Hm `os.makedirs()` is supposed to make any directories that are missing in that path. Howdy 2 should always have write permission to /security/howdy, so maybe the permissions for Howdy are incorrect

yougotborked, 2023-10-22:
Just wanted to add that I was experiencing this problem too. Manually creating the dir fixed this for me on fedora 39 beta. 

I originally configured howdy in Fedora using the script referenced in the wiki
https://gist.github.com/m1nicrusher/35e79b20553c8863e0c642f8d801da7f

saschaeggi, 2023-11-02:
@tokox thank you, that was the root of the problem on my end as well (Fedora 38)

armeldemarsac92, 2023-11-27:
fixed it for me too ! thanks a lot man

musikid, 2023-12-23:
Adding an entry to the wiki with the solution, thanks  @tokox.

closed 2023-12-23 by musikid