boltgolt/howdy #804: Howdy not working on lockscreen on Fedora 38

SoyBoy10, 2023-06-23:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

I recently installed Fedora 38 workstation on my computer and wanted to enable facial recognition to authenticate. I configured howdy using the following tutorial : https://copr.fedorainfracloud.org/coprs/principis/howdy/ 
First thing that I don't really understand is that i don't have a pam.d/kde file. My computer seemed to be using gdm so i change gdm-password by adding the 2 first lines and know it look like that :
[maxime@laptop-hij2thu7 pam.d]$ cat gdm-password 
auth     [success=done ignore=ignore default=bad] pam_selinux_permit.so
auth        sufficient    pam_python.so /lib64/security/howdy/pam.py
auth        substack      password-auth
auth        optional      pam_gnome_keyring.so
auth        include       postlogin

account     required      pam_nologin.so
account     include       password-auth

password    substack       password-auth
-password   optional       pam_gnome_keyring.so use_authtok

session     required      pam_selinux.so close
session     required      pam_loginuid.so
session     required      pam_selinux.so open
session     optional      pam_keyinit.so force revoke
session     required      pam_namespace.so
session     include       password-auth
session     optional      pam_gnome_keyring.so auto_start
session     include       postlogin

I didn't change the sudo file because i like to be typing my password in terminal but only wanted facial auth to unlock my computer. With the following configutation i got facial auth working on boot, but when i just lock my computer i got Unknown Error 1. I checked the log of my computer and got the 2 following error :
error 1:
ImportError: /home/maxime/.local/lib/python3.11/site-packages/_dlib_pybind11.cpython-311-x86_64-linux-gnu.so: failed to map segment from shared object
     from _dlib_pybind11 import *
AVC avc:  denied  { execute } for  pid=4550 comm="python3" path="/home/maxime/.local/lib/python3.11/site-packages/_dlib_pybind11.cpython-311-x86_64-linux-gnu.so" dev="nvme0n1p3" ino=9242 scontext=system_u:system_r:xdm_t:s0-s0:c0.c1023 tcontext=unconfined_u:object_r:gconf_home_t:s0 tclass=file permissive=0

error 2:
Deleting problem directory '/var/spool/abrt/Python3-2023-06-23-14:06:19-4550'
'post-create' on '/var/spool/abrt/Python3-2023-06-23-14:06:19-4550' exited with 1
Package 'howdy' isn't signed with proper key

I think my problem is more related to the first error that shows permission error. I also tried to setenforce 0 end with SELinux disable the facial work even when screen is locked. But since having SELinux disable is not a good practice i guess, i would prefer to fix this permission issue. In addition to that i added a howdy.te file as follow:
[maxime@laptop-hij2thu7 ~]$ cat howdy.te
module howdy 1.0;

require {
    type lib_t;
    type xdm_t;
    type v4l_device_t;
    type sysctl_vm_t;
    class chr_file map;
    class file { create getattr open read write };
    class dir add_name;
}

#============= xdm_t ==============
allow xdm_t lib_t:dir add_name;
allow xdm_t lib_t:file { create write };
allow xdm_t sysctl_vm_t:file { getattr open read };
allow xdm_t v4l_device_t:chr_file map;

and did the following commands :
checkmodule -M -m -o howdy.mod howdy.te
semodule_package -o howdy.pp -m howdy.mod
semodule -i howdy.pp

I also changef various permission settings of directory. Thanks in advance for the help
 

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Fedora 38 Workstation

Howdy version (`sudo howdy version`):
Howdy 2.6.0

SoyBoy10, 2023-06-26:
# Update
I found how to fix my problem following the guide here : https://linuxreviews.org/Howdy/SELinux

I have a slightly different howdy.te file but the idea is the same:

[maxime@laptop-hij2thu7 ~]$ cat howdy.te
module howdy 1.0;

require {
    type lib_t;
    type xdm_t;
    type v4l_device_t;
    type sysctl_vm_t;
    type gconf_home_t;
    class chr_file map;
    class file { execute create getattr open read write };
    class dir add_name;
}

#============= xdm_t ==============
allow xdm_t lib_t:dir add_name;
allow xdm_t gconf_home_t:file execute;
allow xdm_t lib_t:file { create write };
allow xdm_t sysctl_vm_t:file { getattr open read };
allow xdm_t v4l_device_t:chr_file map;

hope it will help :)

closed 2023-06-26 by SoyBoy10

mavinii, 2023-11-13:
> # Update
> I found how to fix my problem following the guide here : https://linuxreviews.org/Howdy/SELinux
>
> I have a slightly different howdy.te file but the idea is the same:
>
> [maxime@laptop-hij2thu7 [...]

I can confirm it worked!

Using the code you provided, it did not work until I recompiled it. I recompiled it by doing:

checkmodule -M -m -o howdy.mod howdy.te
semodule_package -o howdy.pp -m howdy.mod
sudo semodule -i howdy.pp

hope it will help too :)

paulo-evangelista, 2024-03-07:
Answer by @maximerochat and @mavinii still working flawlessly on fedora 39 with the newest versions of Howdy. Thanks you guys!

paulo-evangelista, 2024-03-07:
Answer by @maximerochat and @mavinii still working flawlessly. Thanks guys!

Dinico414, 2024-05-19:
Didn't help me sadly