boltgolt/howdy #805: Howdy keeps looping and locked me out of sudo

nemkicsiaron, 2023-06-24:
After a fresh Neon install I couldn't get the (infrared) cameras to work to register a model, thus I tried the KDE Plasma [workaround detailed in it's respective Wiki page](https://github.com/boltgolt/howdy/wiki/Howdy-on-Debian-Testing---KDE-Plasma#fix-the-script-for-debian-testing).

After that it kept giving me Unknown Error 1 until it hits the timeout and the 3 time sudo password guess where the auth process ends. Hence I don't even get a chance to enter my password like normally. I am no longer able to authenticate as an admin.

Can anyone help me with at least removing howdy at first, and potentially figure out what went wrong?

Here are my relevant files, that I figured would be needed:

<details><summary>config.ini</summary>
<p>
<code>
[core]
detection_notice = false
no_confirmation = true
suppress_unknown = false
ignore_ssh = true
ignore_closed_lid = true
disabled = false
use_cnn = false
[video]
certainty = 2.8
timeout = 4
device_path = /dev/video0
max_height = 320
frame_width = -1
frame_height = -1
dark_threshold = 100
recording_plugin = opencv
device_format = v4l2
force_mjpeg = false
exposure = -1
[snapshots]
capture_failed = true
capture_successful = true
[debug]
end_report = true
</code>
</p>
</details> 

<details><summary>auth.log</summary>
<code>
Jun  4 00:01:48 zenbook [HOWDY]: Attempting facial authentication for user [username]
Jun  4 00:01:48 zenbook [HOWDY]: Failure, unknown error1
Jun  4 00:17:01 zenbook CRON[5249]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jun  4 00:17:01 zenbook CRON[5249]: pam_unix(cron:session): session closed for user root
Jun  4 13:38:57 zenbook systemd-logind[702]: Operation 'sleep' finished.
Jun  4 13:39:09 zenbook [HOWDY]: Attempting facial authentication for user [username]
Jun  4 13:39:09 zenbook [HOWDY]: Failure, unknown error1
Jun  4 13:53:13 zenbook [HOWDY]: Attempting facial authentication for user [username]
Jun  4 13:53:13 zenbook [HOWDY]: Failure, unknown error1
Jun  4 13:54:24 zenbook [HOWDY]: Attempting facial authentication for user [username]
Jun  4 13:54:28 zenbook [HOWDY]: Failure, timeout reached
Jun  4 13:54:32 zenbook sudo:   [username] : TTY=pts/1 ; PWD=/home/[username] ; USER=root ; COMMAND=/usr/bin/chmod +x /lib/security/howdy/recorders/~
Jun  4 13:54:32 zenbook sudo: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Jun  4 13:54:32 zenbook sudo: pam_unix(sudo:session): session closed for user root
Jun  4 14:53:52 zenbook systemd-logind[702]: Operation 'sleep' finished.
Jun  4 14:53:53 zenbook [HOWDY]: Attempting facial authentication for user [username]
Jun  4 14:53:53 zenbook [HOWDY]: Failure, unknown error1
Jun  4 14:54:19 zenbook CRON[6249]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jun  4 14:54:19 zenbook CRON[6249]: pam_unix(cron:session): session closed for user root
Jun  4 14:54:55 zenbook [HOWDY]: Attempting facial authentication for user [username]
Jun  4 14:54:59 zenbook [HOWDY]: Failure, timeout reached
Jun  4 14:55:01 zenbook sudo:   [username] : TTY=pts/1 ; PWD=/usr/lib/security/howdy ; USER=root ; COMMAND=/usr/bin/chmod +x recorders
Jun  4 14:55:01 zenbook sudo: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Jun  4 14:55:01 zenbook sudo: pam_unix(sudo:session): session closed for user root
Jun  4 14:55:13 zenbook sudo:   [username] : TTY=pts/1 ; PWD=/usr/lib/security/howdy ; USER=root ; COMMAND=/usr/bin/chmod o+x dlib-data/
Jun  4 14:55:13 zenbook sudo: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Jun  4 14:55:13 zenbook sudo: pam_unix(sudo:session): session closed for user root
Jun  4 15:23:31 zenbook systemd-logind[702]: Operation 'sleep' finished.
[... 419 lines cut ...]
Jun  3 18:20:05 zenbook [HOWDY]: Failure, timeout reached
Jun  3 18:20:07 zenbook sudo:   [username] : TTY=pts/1 ; PWD=/home/[username] ; USER=root ; COMMAND=/usr/bin/tail /var/log/auth.log
Jun  3 18:20:07 zenbook sudo: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Jun  3 18:20:07 zenbook sudo: pam_unix(sudo:session): session closed for user root
Jun  3 18:30:01 zenbook CRON[4290]: pam_unix(cron:session): session opened for user root(uid=0) by (uid=0)
Jun  3 18:30:01 zenbook CRON[4290]: pam_unix(cron:session): session closed for user root
Jun  3 18:40:27 zenbook systemd-logind[702]: Operation 'sleep' finished.
Jun  3 18:40:27 zenbook [HOWDY]: Attempting facial authentication for user [username]
Jun  3 18:40:27 zenbook [HOWDY]: Failure, unknown error1
Jun  3 18:42:01 zenbook [HOWDY]: Attempting facial authentication for user [username]
Jun  3 18:42:06 zenbook [HOWDY]: Failure, timeout reached
Jun  3 18:42:07 zenbook sudo:   [username] : TTY=pts/1 ; PWD=/home/[username] ; USER=root ; COMMAND=/usr/sbin/apt install -y python3-opencv
Jun  3 18:42:07 zenbook sudo: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
Jun  3 18:42:12 zenbook sudo: pam_unix(sudo:session): session closed for user root
Jun  4 00:01:48 zenbook systemd-logind[702]: Operation 'sleep' finished.

</code>
</details> 
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): KDE Neon 5.27 (Ubuntu 22.04)

Howdy version (`sudo howdy version`): cannot

dbolger, 2023-11-18:
Unsure if you discovered a fix for this, but what I had to do was `chroot` into my operating system and revert all changes to `/etc/pam.d/sudo` back to its original state.

cnrenil, 2023-11-21:
```
echo "password" | sudo -S apt purge howdy
```

dbolger, 2023-11-22:
> ```
> echo "password" | sudo -S apt purge howdy
> ```

This doesn't work for operating systems that don't have `apt`. In my case, I'm using Void Linux.

cnrenil, 2023-11-23:
Try to enter the system rescue mode through grub. The specific method can be found on Google

In this mode, you can obtain file access permissions by remounting the root directory

Modify the pam file located in the /etc directory, which may be in /etc/pam.d/common_auth, depending on your system. Delete the line where howdy is located

If you think it's too troublesome, you can try modifying the sudo -S command and using the sed command to delete the relevant lines. There are always ways to solve problems.

--translated by ai

cnrenil, 2023-11-23:
> > ```
> > echo "password" | sudo -S apt purge howdy
> > ```
> 
> This doesn't work for operating systems that don't have `apt`. In my case, I'm using Void Linux.

Try to enter the system rescue mode through grub. The specific method can be found on Google

In this mode, you can obtain file access permissions by remounting the root directory

Modify the pam file located in the /etc directory, which may be in /etc/pam.d/common_auth, depending on your system. Delete the line where howdy is located

If you think it's too troublesome, you can try modifying the sudo -S command and using the sed command to delete the relevant lines. There are always ways to solve problems.

--translated by ai

closed 2024-04-15 by musikid