boltgolt/howdy #807: Debian 12 and externally-managed-environment error message

GillesRonsin, 2023-06-27:
Hi,
I was using Howdy with Debian 11 with happiness..
Since I upgrade to Debian 12, howdy has been disabled
I can no more remove or reinstall howdy
Even trying to disable security 

sudo PIP_BREAK_SYSTEM_PACKAGE=1 apt remove --purge howdy
Après cette opération, 166 ko d'espace disque seront libérés.
Souhaitez-vous continuer ? [O/n] o
(Lecture de la base de données... 248123 fichiers et répertoires déjà installés.)
Suppression de howdy (2.6.1) ...
Can't remove executable
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Traitement des actions différées (« triggers ») pour man-db (2.11.2-2) ...

In case of install again:

sudo PIP_BREAK_SYSTEM_PACKAGE=1 gdebi Téléchargements/howdy_2.6.1.deb
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Reading state information... Done

Howdy: Windows Hello style authentication for Linux.
 Use your built-in IR emitters and camera in combination with face recognition
 to prove who you are.
Voulez-vous installer le paquet logiciel ? [o/N] :o
/usr/bin/gdebi:113: FutureWarning: Possible nested set at position 1
  c = findall("[[(](\S+)/\S+[])]", msg)[0].lower()
Sélection du paquet howdy précédemment désélectionné.
(Lecture de la base de données... 248088 fichiers et répertoires déjà installés.)
Préparation du dépaquetage de .../howdy_2.6.1.deb ...

WARNING: Could not automatically find the right webcam, manual configuration after installation required

Starting certainty auto config...

After detection, Howdy knows how certain it is that the match is correct.
How certain Howdy needs to be before authenticating you can be customized.

F: Fast.
Allows more fuzzy matches, but speeds up the scanning process greatly.

B: Balanced.
Still relatively quick detection, but might not log you in when further away.

S: Secure.
The safest option, but will take much longer to authenticate you.

You can always change this setting in the config.
What profile would you like to use? [f/b/s]: b

Dépaquetage de howdy (2.6.1) ...
Paramétrage de howdy (2.6.1) ...

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Deban 12

Howdy version (`sudo howdy version`): was 2.6.1

mathieujobin, 2023-06-28:
I had a similar issue with Kubuntu 23.04 (latest non-LTS at the moment)

```
Setting up libopencv-dev (4.6.0+dfsg-11) ...
Setting up howdy (2.6.1) ...

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.11.2-1) ...
Processing triggers for libc-bin (2.37-0ubuntu2) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

just trying again actually it passed
```
$ sudo apt install howdy
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
howdy is already the newest version (2.6.1).
The following packages were automatically installed and are no longer required:
  python3-awscrt python3-jmespath python3-prompt-toolkit python3-pyasn1 python3-wcwidth
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
1 not fully installed or removed.
After this operation, 0 B of additional disk space will be used.
Do you want to continue? [Y/n] 
Setting up howdy (2.6.1) ...
Attempting installation of missing data files
Downloading 3 required data files...
dlib_face_recognition_resnet_model_v1.d 100%[============================================================================>]  20.44M  6.21MB/s    in 3.5s    
mmod_human_face_detector.dat.bz2        100%[============================================================================>] 678.41K  3.49MB/s    in 0.2s    
shape_predictor_5_face_landmarks.dat.bz 100%[============================================================================>]   5.44M  3.21MB/s    in 1.7s    
Unpacking...
```

but It does not provide a `howdy` executable

```
$ sudo howdy add
sudo: howdy: command not found
$ dpkg -l | grep howdy
ii  howdy                                            2.6.1                                       all          Howdy: Windows Hello style authentication for Linux.
$ dpkg -L howdy
/.
/lib
/lib/security
/lib/security/howdy
/lib/security/howdy/recorders
/lib/security/howdy/recorders/ffmpeg_reader.py
/lib/security/howdy/recorders/pyv4l2_reader.py
/lib/security/howdy/recorders/v4l2.py
/lib/security/howdy/recorders/video_capture.py
/lib/security/howdy/recorders/__init__.py
/lib/security/howdy/pam.py
/lib/security/howdy/logo.png
/lib/security/howdy/cli.py
/lib/security/howdy/dlib-data
/lib/security/howdy/dlib-data/Readme.md
/lib/security/howdy/dlib-data/install.sh
/lib/security/howdy/compare.py
/lib/security/howdy/cli
/lib/security/howdy/cli/add.py
/lib/security/howdy/cli/test.py
/lib/security/howdy/cli/list.py
/lib/security/howdy/cli/disable.py
/lib/security/howdy/cli/__init__.py
/lib/security/howdy/cli/config.py
/lib/security/howdy/cli/snap.py
/lib/security/howdy/cli/clear.py
/lib/security/howdy/cli/remove.py
/lib/security/howdy/snapshot.py
/lib/security/howdy/pam-config
/lib/security/howdy/pam-config/howdy
/lib/security/howdy/config.ini
/usr
/usr/share
/usr/share/doc
/usr/share/doc/howdy
/usr/share/doc/howdy/changelog.gz
/usr/share/doc/howdy/copyright
/usr/share/man
/usr/share/man/man1
/usr/share/man/man1/howdy.1.gz
/usr/share/bash-completion
/usr/share/bash-completion/completions
/usr/share/bash-completion/completions/howdy
/usr/share/pam-configs
/usr/share/pam-configs/howdy
/usr/share/lintian
/usr/share/lintian/overrides
/usr/share/lintian/overrides/howdy
```

sherrmann, 2023-07-01:
The link to the executable was missing on my kubuntu install as well.

Solution:
1: apt install howdy 
2: apt install howdy 
3: sudo ln /lib/security/howdy/cli.py /usr/local/bin/howdy

Credits to mathieujobin and https://github.com/boltgolt/howdy/issues/771#issuecomment-1491185055

GabrielVento, 2023-07-19:
Same issue here. I manage to install Howdy and create the link to de exec file, but no hodwy command can be execute. The problem seen to be the cli module, but I can´t fix it. Here the issue I'm facing:

user@computer:/lib/security/howdy$ sudo howdy -U user test
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 109, in <module>
    import cli.test
ModuleNotFoundError: No module named 'cli'

The same error I receive with any other Howdy command.

I was using Howdy with no issues in Debian 11. Since I upgrade to Debian 12 it stopped working.

Debian Version: 12.0
Howdy version: 2.6.1 (from the .deb package specific for Debian distro)

Thank you in advance!

Ra72xx, 2023-07-22:
Same here: Anybody found a solution yet?

gllrmzndm, 2023-08-07:
> Same here on Ubuntu 23.04: Anybody found a solution yet?

`sudo apt install python3-full`
`sudo apt install python3-pip
`sudo apt install python3-venv`

`cd env`
`source bin/activate`

`pip install -r requirements.txt --break-system-packages`

Ra72xx, 2023-08-07:
I managed to install it somehow this weekend by installing libjpeg62-turbo-12.1.5-2 and python3-dlib-19.24.2-1 from some older Debian (or Ubuntu?) repository.

stucash, 2023-08-20:
I hope the team could roll out an update soon to incorporate `pipx` to install howdy; that's the recommended way from python official team; I am not sure if this is going to work for howdy but it seems that the error message from pip is suggesting that `pipx` is going to manage a vn for us and we should be able to use the application hassle-free.

shorenster, 2024-04-27:
Same issue here, debian 12, howdy 2.6.1 someone found a solution?

symumsakin, 2024-05-18:
facing same issue

DaanSelen, 2025-02-24:
The same issue is still present.

theogayar, 2025-05-16:
Issue still present today, trying to install on Kubuntu with Ubuntu 25.04

- ran twice "sudo apt install howdy", and got a python error first time around 
- no executable, fixed by "sudo ln /lib/security/howdy/cli.py /usr/local/bin/howdy", as suggested per #https://github.com/boltgolt/howdy/issues/771#issuecomment-1491185055
- now getting a "no module named cli" error

SCI-co-Newb, 2025-06-02:
Instead of linking it, I used /lib/security/howdy/cli.py as howdy, and that got me past the 'no module named cli' part.

soraxas, 2025-09-10:
> The link to the executable was missing on my kubuntu install as well.
> 
> Solution: 1: apt install howdy 2: apt install howdy 3: sudo ln /lib/security/howdy/cli.py /usr/local/bin/howdy
> 
> Credits to mathieujobin and [#771 (comment)](https://github.com/boltgolt/howdy/issues/771#issuecomment-1491185055)

You should do a symlink instead of hardlink. 

I.e.
```sh
sudo ln -s /lib/security/howdy/cli.py /usr/local/bin/howdy
```

zahlman, 2025-12-14:
The issue isn't going to go away as long as the project is packaged the way it is. Right now it's trying to use pip to install some packages in the system environment (rather: install OpenCV if missing, and upgrade pip and NumPy). It also tries to build [another project](https://github.com/davisking/dlib/) from source. All of this is beyond what I'd expect Apt to be able to cope with, which is [why the externally-managed-environment error message exists](https://peps.python.org/pep-0668/) in the first place. Especially since right now it's trying to upgrade these things to whatever is the current version, without any check for compatibility. That's fine for many projects and it may well be fine for this one, but it's not how things are expected to work in the system environment of a point-release distro.

Not every Debian (or downstream of that) user even *has* a system installation of pip in the first place (as this is packaged separately from the base Python on Debian); pip cannot self-upgrade when absent.

The general expectation in the Python ecosystem, meanwhile, is to [use native Python packaging](https://packaging.python.org/). I understand that the project has C code that needs to be built from source, and that the dlib dependency also needs to be built from source. But Python's packaging system accommodates that, and the source package for dlib is available on PyPI. So a hypothetical "sdist" for this project would simply declare NumPy, OpenCV and dlib (with appropriate versions) as dependencies.

Running the install twice, symlinking missing pieces etc. are hacks to work around the fact that the post-install script didn't complete. Post-install scripts aren't supposed to have errors like this. The installation process described by this post-install is fundamentally unsound; even with a hard requirement to stick within the system environment, that should also mean sticking within an Apt ecosystem — by using Apt packages for Numpy and OpenCV (and of course, this way pip becomes irrelevant).

closed 2026-11-16 by boltgolt