boltgolt/howdy #810: qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.

phil-lipp, 2023-06-30:
I get the following error when running `sudo howdy test` in `su -, however this error doesn't happen when running in `su`. I think I might have screwed during the installation process...
Is there a way to fix this?

```
qt.qpa.xcb: could not connect to display 
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.
```

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version (`sudo howdy version`):

fabmorais, 2023-07-05:
@phil-lipp if you're running Wayland, please check this issue: https://github.com/boltgolt/howdy/issues/104.

I had a similar problem (arch+wayland) and the suggestion from @sapjunior helped me to fix it.

Hope it help/fix yours too 🚀

TurtleMeds, 2025-05-20:
for anyone wanting the full solution:

1. Install xhost:
- for Ubuntu
    ```bash
    sudo apt install xhost
    ```
 - For Arch:
      ```
      sudo pacman -S xhost
      ```

2. Put this file somewhere in your path (i recommend ~/.local/bin/wsudo):

```bash
#!/bin/bash
#small script to enable root access to x-windows system 
xhost +SI:localuser:root
sudo $1
#disable root access after application terminates
xhost -SI:localuser:root
#print access status to allow verification that root access was removed
xhost
```
3. Add execution permissions:
```bash
chmod +x ~/.local/bin/wsudo
```
4. and run:

```bash
wsudo howdy test
```
to start the test.

closed 2026-11-16 by boltgolt