boltgolt/howdy #811: Login loop after face recognition

Ardenos, 2023-07-02:
Hello,

I was a big user of Howdy some months ago and now, I had to encrypt my whole disk for my work (so I had to format it). I tried to install Howdy again, everything went fine: I added my face, tried to make it work with a random sudo command (which works).

But the problem occurs when I boot up my computer : as I said before, I encrypted my PC with LVM and my home folder, so when I boot up my PC, it asks me for the encryption key to open Mint, I write it then it leads me to the Mint login screen, where my camera blinks, it recognizes my face and when I press "Log In", it puts a black screen then comes back to the login screen and it loops on the camera flashing again. To login, I have to cover my camera with my hand and then type my password to enter.

I tried several thing to make it work, like changing the order in /etc/pam.d/common-auth or in the same file, changing the auth method of Howdy to forward_pass (I don't know exactly what it means). I also tried to put 'no_confirmation' to true, but it doesn't change anything.

Here is the content of my log file in /var/log/auth.log when it recognizes my face :
Jul  2 11:55:02 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[955]: System is rebooting.
Jul  2 11:55:31 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[943]: New seat seat0.
Jul  2 11:55:31 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[943]: Watching system buttons on /dev/input/event2 (Power Button)
Jul  2 11:55:31 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[943]: Watching system buttons on /dev/input/event0 (Lid Switch)
Jul  2 11:55:31 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[943]: Watching system buttons on /dev/input/event1 (Power Button)
Jul  2 11:55:31 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[943]: Watching system buttons on /dev/input/event10 (ASUE140A:00 04F3:3134 Keyboard)
Jul  2 11:55:31 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[943]: Watching system buttons on /dev/input/event3 (AT Translated Set 2 keyboard)
Jul  2 11:55:32 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_unix(lightdm-greeter:session): session opened for user lightdm(uid=112) by (uid=0)
Jul  2 11:55:32 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[943]: New session c1 of user lightdm.
Jul  2 11:55:32 ardenos-ZenBook-UX325EA-UX325EA systemd: pam_unix(systemd-user:session): session opened for user lightdm(uid=112) by (uid=0)
Jul  2 11:55:32 ardenos-ZenBook-UX325EA-UX325EA lightdm: gkr-pam: gnome-keyring-daemon started properly
Jul  2 11:55:32 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_succeed_if(lightdm:auth): requirement "user ingroup nopasswdlogin" not met by user "ardenos"
Jul  2 11:55:42 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_succeed_if(lightdm:auth): requirement "user ingroup nopasswdlogin" not met by user "ardenos"
Jul  2 11:55:50 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_succeed_if(lightdm:auth): requirement "user ingroup nopasswdlogin" not met by user "ardenos"
Jul  2 11:56:02 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_succeed_if(lightdm:auth): requirement "user ingroup nopasswdlogin" not met by user "ardenos"
Jul  2 11:56:19 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_unix(lightdm:auth): authentication failure; logname= uid=0 euid=0 tty=:0 ruser= rhost=  user=ardenos
Jul  2 11:56:19 ardenos-ZenBook-UX325EA-UX325EA [HOWDY]: Attempting facial authentication for user ardenos
Jul  2 11:56:20 ardenos-ZenBook-UX325EA-UX325EA [HOWDY]: Login approved
Jul  2 11:56:20 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_ecryptfs: Passphrase file wrapped
Jul  2 11:56:20 ardenos-ZenBook-UX325EA-UX325EA lightdm: Incorrect wrapping key for file [/home/ardenos/.ecryptfs/wrapped-passphrase]
Jul  2 11:56:20 ardenos-ZenBook-UX325EA-UX325EA lightdm: Error attempting to unwrap passphrase from file [/home/ardenos/.ecryptfs/wrapped-passphrase]; rc = [-5]
Jul  2 11:56:20 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_ecryptfs: Error adding passphrase key token to user session keyring; rc = [-5]
Jul  2 11:56:20 ardenos-ZenBook-UX325EA-UX325EA lightdm: gkr-pam: unable to locate daemon control file
Jul  2 11:56:20 ardenos-ZenBook-UX325EA-UX325EA lightdm: gkr-pam: stashed password to try later in open session
Jul  2 11:56:21 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_unix(lightdm-greeter:session): session closed for user lightdm
Jul  2 11:56:21 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_unix(lightdm:session): session opened for user ardenos(uid=1000) by (uid=0)
Jul  2 11:56:21 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[943]: Removed session c1.
Jul  2 11:56:21 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[943]: New session c2 of user ardenos.
Jul  2 11:56:21 ardenos-ZenBook-UX325EA-UX325EA systemd: pam_unix(systemd-user:session): session opened for user ardenos(uid=1000) by (uid=0)
Jul  2 11:56:21 ardenos-ZenBook-UX325EA-UX325EA lightdm: gkr-pam: gnome-keyring-daemon started properly and unlocked keyring
Jul  2 11:56:21 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[943]: Session c2 logged out. Waiting for processes to exit.
Jul  2 11:56:22 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_unix(lightdm-greeter:session): session opened for user lightdm(uid=112) by (uid=0)
Jul  2 11:56:22 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[943]: New session c3 of user lightdm.
Jul  2 11:56:22 ardenos-ZenBook-UX325EA-UX325EA systemd-logind[943]: Removed session c2.
Jul  2 11:56:22 ardenos-ZenBook-UX325EA-UX325EA lightdm: gkr-pam: gnome-keyring-daemon started properly
Jul  2 11:56:23 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_succeed_if(lightdm:auth): requirement "user ingroup nopasswdlogin" not met by user "ardenos"
Jul  2 11:56:33 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_succeed_if(lightdm:auth): requirement "user ingroup nopasswdlogin" not met by user "ardenos"
Jul  2 11:56:35 ardenos-ZenBook-UX325EA-UX325EA lightdm: pam_unix(lightdm:auth): authentication failure; logname= uid=0 euid=0 tty=:0 ruser= rhost=  user=ardenos

Hopes someone can help me with this,

Best regards, Ardenos
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Linux Mint 

Howdy version (`sudo howdy version`): 2.6.1

lug-gh, 2023-07-07:
Having the same issue (although I use elemtary OS - which is Ubuntu based too).

Sometime it works, sometimes it doesn't, but auth.log is not giving any useful information.

`unsuccesful login`
```
Jul  7 12:36:15 dell lightdm: pam_unix(lightdm-greeter:session): session opened for user lightdm(uid=115) by (uid=0)
Jul  7 12:36:15 dell systemd-logind[874]: New session c4 of user lightdm.
Jul  7 12:36:16 dell systemd: pam_unix(systemd-user:session): session opened for user lightdm(uid=115) by (uid=0)
Jul  7 12:36:16 dell lightdm: gkr-pam: gnome-keyring-daemon started properly
Jul  7 12:36:16 dell lightdm: pam_succeed_if(lightdm:auth): requirement "user ingroup nopasswdlogin" not met by user "lug-gh"
Jul  7 12:36:16 dell [HOWDY]: Attempting facial authentication for user lug-gh
Jul  7 12:36:17 dell lightdm: pam_succeed_if(lightdm:auth): requirement "user ingroup nopasswdlogin" not met by user "lug-gh"
Jul  7 12:36:17 dell [HOWDY]: Attempting facial authentication for user lug-gh
Jul  7 12:36:18 dell [HOWDY]: Failure, unknown error1
Jul  7 12:36:22 dell lightdm: gkr-pam: unable to locate daemon control file
Jul  7 12:36:22 dell lightdm: gkr-pam: stashed password to try later in open session
Jul  7 12:36:22 dell systemd-logind[874]: Removed session c4.
```

`succesful login`
```
Jul  7 14:32:12 dell lightdm: pam_unix(lightdm-greeter:session): session opened for user lightdm(uid=115) by (uid=0)
Jul  7 14:32:12 dell systemd-logind[874]: New session c6 of user lightdm.
Jul  7 14:32:12 dell systemd: pam_unix(systemd-user:session): session opened for user lightdm(uid=115) by (uid=0)
Jul  7 14:32:12 dell lightdm: gkr-pam: gnome-keyring-daemon started properly
Jul  7 14:32:13 dell lightdm: pam_succeed_if(lightdm:auth): requirement "user ingroup nopasswdlogin" not met by user "lug-gh"
Jul  7 14:32:13 dell [HOWDY]: Attempting facial authentication for user lug-gh
Jul  7 14:32:14 dell [HOWDY]: Login approved
Jul  7 14:32:14 dell lightdm: gkr-pam: no password is available for user
Jul  7 14:32:14 dell systemd-logind[874]: Removed session c6
```

Kad001, 2024-05-18:
Same issue here 

Distro: Linux Mint
Howdy version: Howdy 2.6.1

I cannot get to the desktop from the login screen.
Howdy scans, detects my face and then i see the option to login. When i enter it loops back and authenticates again.
So the lock screen does not unlock.
I set the "no_notification to true" so it no longer states who is logged in but the login loop is still there.

However, once I use my password to login, howdy is ok for sudo tasks

Did you guys find a solution?

Ardenos, 2024-05-18:
Hey, no I didn't find anything, I just stopped using it. But the last time I was using it I had the same behavior as the one you described. If you ever find something, I'm still up for it

rabsbustamante, 2024-05-29:
I'm having exactly the same issue and can't find a solution for it. Here are my specs:

System:
  Kernel: 5.15.0-107-generic x86_64 bits: 64 compiler: gcc v: 11.4.0 Desktop: MATE 1.26.0
    info: mate-panel wm: marco 1.26.0 dm: LightDM 1.30.0 Distro: Linux Mint 21.3 Virginia
    base: Ubuntu 22.04 jammy
Machine:
  Type: Laptop System: Dell product: Latitude 5490 v: N/A serial: <superuser required> Chassis:
    type: 10 serial: <superuser required>
  Mobo: Dell model: 0DH2HV v: A00 serial: <superuser required> UEFI: Dell v: 1.33.0
    date: 12/12/2023
CPU:
  Info: quad core model: Intel Core i5-8350U bits: 64 type: MT MCP smt: enabled arch: Coffee Lake
    rev: A cache: L1: 256 KiB L2: 1024 KiB L3: 6 MiB
  Speed (MHz): avg: 2997 high: 3001 min/max: 400/3600 cores: 1: 3000 2: 3000 3: 3001 4: 3000
    5: 3000 6: 3000 7: 2979 8: 3000 bogomips: 30399
  Flags: avx avx2 ht lm nx pae sse sse2 sse3 sse4_1 sse4_2 ssse3 vmx
Graphics:
  Device-1: Intel UHD Graphics 620 vendor: Dell driver: i915 v: kernel ports: active: eDP-1
    empty: DP-1,HDMI-A-1,HDMI-A-2 bus-ID: 00:02.0 chip-ID: 8086:5917 class-ID: 0300
  Device-2: Realtek Integrated_Webcam_HD type: USB driver: uvcvideo bus-ID: 1-5:3
    chip-ID: 0bda:58f6 class-ID: 0e02 serial: <filter>
  Display: x11 server: X.Org v: 1.21.1.4 compositor: marco v: 1.26.0 driver: X:
    loaded: modesetting unloaded: fbdev,vesa gpu: i915 display-ID: :0 screens: 1
  Screen-1: 0 s-res: 1600x900 s-dpi: 96 s-size: 423x238mm (16.7x9.4") s-diag: 485mm (19.1")
  Monitor-1: eDP-1 model: BOE Display res: 1600x900 hz: 60 dpi: 132 size: 309x173mm (12.2x6.8")
    diag: 354mm (13.9") modes: 1920x1080
  OpenGL: renderer: Mesa Intel UHD Graphics 620 (KBL GT2) v: 4.6 Mesa 23.2.1-1ubuntu3.1~22.04.2
    direct render: Yes

Kad001, 2024-05-29:
Hi, I found that this issue occurs when you encrypt the home folder of your installation. Only the password works to get you in. 
Once you remove the home folder encryption Howdy is able to sign you in. I tested this on both a dell latitude 5400 and Asus Zenbook Pro Duo with the same results. 
So you can either remove the home drive encryption / do a fresh install without encrypting the home folder or use a password to sign in. Which comes down to which is more important for you.
Note that if Howdy logs you into your desktop, you still need a password to unlock your keyring. 
Hope this helps.

rabsbustamante, 2024-05-30:
You're right, I have my home folder encrypted. I guess I'd continue using howdy and covering the camera when loggin in, I actually like not type my password every time I write a sudo XD.

lug-gh, 2024-05-30:
> Hi, I found that this issue occurs when you encrypt the home folder of your installation. Only the password works to get you in. Once you remove the home folder encryption Howdy is able to sign you [...]

for me it happens without encryption