boltgolt/howdy #813: 3.0.0 BETA .deb file for Debian

phil-lipp, 2023-07-04:
I've seen that for some distros (e.g. Arch) the 3.0.0 BETA is already available. Is it possible to provide a .deb of the BETA for the Debian users?

Reason being that 2.6.1 has a couple of issues that as far as I can see are already addressed in some commits

boltgolt, 2023-07-09:
You could just download them from Launchpad directly, the packages should have an amd64 .deb when you expand them: https://launchpad.net/~boltgolt/+archive/ubuntu/howdy-beta/+packages

rayzorben, 2023-11-13:
Seems like it hasn't been updated since February. Is there a way to request an update, or are there instructions for building the beta on Ubuntu 23.10 ?

rayzorben, 2023-12-21:
@boltgolt sorry to bother, i'm willing to build this myself on Ubuntu 23.10 and get it installed. The beta seemed to work better for me on Arch so that is what I am hoping to use. Do you have any instructions on  how to build, tools required, etc.? Or a build script?

musikid, 2023-12-23:
I added a section to build from source in the README, and as for the package a new build will be done when it's possible.

closed 2023-12-23 by musikid