boltgolt/howdy #815: openSUSE Tumbleweed - howdy doesn't launch at login/boot, sudo howdy test returns errors

Ultome, 2023-07-15:
_Please describe the issue in as much detail as possible, including any errors and traces._

After installing howdy and adding `auth    sufficient      pam_python.so /usr/lib64/security/howdy/pam.py` before `auth    required        pam_unix.so     try_first_pass` in `/etc/pam.d/common-auth` (symlink deleted and hard copy of `common-auth-pc` created instead), I have two problems:

1.  though `sudo howdy --user myusername add` works fine and I can save faces, `sudo howdy test` fails. Here's the output of that command:
```
[ WARN:0@0.237] OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.238] OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.238] OpenCV | GStreamer warning: GStreamer: pipeline have not been created

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.
```
I tried reinstalling xcb in different ways (ended up breaking my system at some point actually), it didn't help.
If I edit `/usr/bin/howdy` so as to bypass root privileges enforcement (even though I'm guessing that's not a good idea for security), `howdy test` run as `$my-non-root-user` works (the video shows up and my face is circled in red), but at any rate, that doesn't help with the second problem:

2. howdy has *never* shown up, nor at boot, logout/login, or screenlocking, no matter the adjustments I tried in PAM config or anywhere else. My webcam (that, I checked, works fine (as mentioned above)) just stays dead in all cases when howdy should kick in except manual calls directly from a terminal.

So I don't know if I missed something is PAM config, howdy install or somewhere else. That's why I need help.

Here is additionnal info that may help:

- output of `ls /etc/pam.d/`:
  ```
common-account     common-password-pc          postlogin-account     postlogin-password-pc
common-account-pc  common-session              postlogin-account-pc  postlogin-session
common-auth        common-session-nonlogin     postlogin-auth        postlogin-session-pc
common-auth-pc     common-session-nonlogin-pc  postlogin-auth-pc     smtp
common-password    common-session-pc           postlogin-password
```
- output of `cat /etc/pam.d/common-auth`:
```
#%PAM-1.0
# a lot of comments...
# ...

auth    required        pam_env.so
auth    optional        pam_kwallet5.so
auth    sufficient      pam_python.so /usr/lib64/security/howdy/pam.py
auth    required        pam_unix.so     try_first_pass
```

Finally, I had to run `sudo pip3 install dlib` because `pip3 install dlib` didn't make the error  about that module disappear during installation, and got a warning about running `pip` with `sudo`.
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution ioenSUSE Tumbleweed (everything including python and pip is up to date — I used `pip3-3.11` for the install):

Howdy version: 2.6.1

Ultome, 2023-07-24:
> You can use `sudo -E howdy test` to run the test, and it has nothing to do with qt and xcb.

Maybe, but the real problem is that howdy doesn't launch at all at login...

Jackaed, 2023-08-10:
The issue is that pam_python uses python3 instead of the python2 that the release version of PAM uses. You can make certain adjustments to correct this and it will work properly. These can be found in other issues.

Ultome, 2023-08-10:
If that was the problem it could be easy to fix, could you please provide links to the issues you're talking about? (there are a lot...)

Jackaed, 2023-08-11:
Change your common-auth to replace pam_python with pam_python3. Then, change pam.py to read "import configparser" instead of "import ConfigParser" and change "config = ConfigParser.ConfigParser()" to "config = configparser.ConfigParser".

Jackaed, 2023-08-11:
I would also recommend using pam includes to include the contents of "common-auth-pc" in "common-auth" instead of just hard copying it, then adding pam in the next line. I don't have the laptop to hand that I did this on, so I can't remember the syntax,

Ultome, 2023-08-11:
I've looked but I didn't find it... (sorry if I'm double-posting but I can't see my comment from yesterday...). Could you please link a thread wtih what I'm looking for? (I also tried to do my own research but I don't really understand what needs which version of Python, and why...)

Cris70, 2023-09-13:
@Jackaed did you find out how to include the contents of "common-auth-pc" in "common-auth"?
I'm interested, too, but haven't been able to find info on the web.

poltpolt, 2024-01-16:
> Change your common-auth to replace pam_python with pam_python3. Then, change pam.py to read "import configparser" instead of "import ConfigParser" and change "config = ConfigParser.ConfigParser()" to "config = configparser.ConfigParser".

for me this helped and all works as expected
in the meantime to get rid of xcb errors I installed
libqt5-dxcbplugin  

instead of sudo howdy i use kdesu howdy and now it shows

closed 2026-11-16 by boltgolt