boltgolt/howdy #816: Can't invoke Howdy with `sudo` but can with `su`.

rokejulianlockhart, 2023-07-16:
I can't invoke Howdy with `sudo howdy test` but can with `su -c 'howdy test'`:

```log
PS /home/rokejulianlockhart> howdy test
Please run this command as root:

        sudo howdy test
NativeCommandExitException: Program "howdy" ended with non-zero exit code: 1.
```

doesn't work, as expected.

```log
PS /home/rokejulianlockhart> sudo howdy test      
[sudo] password for root: 
[ WARN:0@0.762] OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.762] OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.762] OpenCV | GStreamer warning: GStreamer: pipeline have not been created

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vnc, wayland-egl, wayland, wayland-xcomposite-egl, wayland-xcomposite-glx, xcb.

NativeCommandExitException: Program "sudo" ended with non-zero exit code: 1.
```

doesn't work either.

```log
PS /home/rokejulianlockhart> su
Password: 
RQN6C6:/home/rokejulianlockhart # howdy test
[ WARN:0@0.079] OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.079] OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.079] OpenCV | GStreamer warning: GStreamer: pipeline have not been created

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

QStandardPaths: runtime directory '/run/user/1000' is not owned by UID 0, but a directory permissions 0700 owned by UID 1000 GID 1000
QSettings::value: Empty key passed
QStandardPaths: runtime directory '/run/user/1000' is not owned by UID 0, but a directory permissions 0700 owned by UID 1000 GID 1000
QSettings::value: Empty key passed
^C
Closing window
RQN6C6:/home/rokejulianlockhart #
```

works.

Linux distribution (if applicable):

```log
PS /home/rokejulianlockhart> uname -a
Linux RQN6C6 6.4.2-1-default #1 SMP PREEMPT_DYNAMIC Mon Jul 10 10:03:25 UTC 2023 (b97b894) x86_64 x86_64 x86_64 GNU/Linux
PS /home/rokejulianlockhart>
```

Howdy version (`sudo howdy version`):

```log
PS /home/rokejulianlockhart> sudo howdy version
Howdy 2.6.1
PS /home/rokejulianlockhart>
```

musikid, 2023-10-02:
Are you using PowerShell as your user shell?

RokeJulianLockhart, 2023-10-02:
Back then, I might have `chsh -s`'d it. Otherwise, I was just using it for most sessions by manually invoking it. Apologies – I don't exactly recall anymore.

musikid, 2023-10-02:
I'm going to close this until you have more information.

closed 2023-10-02 by musikid

beedell-roke, 2026-11-14:
@saidsay-so, what more information was desired? The distinction that I made shouldn't matter.