boltgolt/howdy #821: cap_v4l.cpp:1973 controlInfo VIDEOIO(V4L2:/dev/video0): property unknown is not supported

Thalyos, 2023-07-31:
I'm rather new to linux and just got a new linux laptop, Fedora 38 is installed on it.
After a flawless install using
sudo dnf copr enable luya/howdy
sudo dnf install howdy

I added v4l-utils.
Then `v4l2-ctl --list-devices`
Chicony USB2.0 Camera: Chicony  (usb-0000:00:14.0-7):
	/dev/video0
	/dev/video1
	/dev/media0
I then edited howdy's config file through `sudo howdy config` and added  the camera path:
device_path = /dev/video0

Then I added my face via `sudo howdy add` 3 times just to be sure. And I did see the little light beside the camera go on as it took the snapshots

However, it just won't work
a `sudo -i` will only prompt me to type in my password, so did rebooting.

I did a sudo howdy test and got the title of the post(a windows with the camera livestream would open during the test, seemed to work just fine):
cap_v4l.cpp:1973 controlInfo VIDEOIO(V4L2:/dev/video0): property unknown is not supported

I also tried different recording plugins but to no avail (ffmpeg and pyv4l2). Test couldn't be done with those though as it wouldn't let me

----
I did find a similar issue : https://github.com/boltgolt/howdy/issues/800
However, the solution of user [tokox](https://github.com/tokox) as described in https://github.com/boltgolt/howdy/issues/801 consisting in creating a new directory : `sudo mkdir /usr/lib64/security/howdy/snapshot` didn't work for me.

And so here I am, any help would be greatly appreciated, a copy of my config file in txt format is attached  is attached below

Linux distribution: Fedora 38

Howdy version 2.6.1
[config.txt](https://github.com/boltgolt/howdy/files/12217512/config.txt)

tokox, 2023-08-16:
>  cap_v4l.cpp:1973 controlInfo VIDEOIO(V4L2:/dev/video0): property unknown is not supported

This is not the issue, so please rename it. I have same thing and everything works.
Did you use [this script](https://gist.github.com/m1nicrusher/35e79b20553c8863e0c642f8d801da7f) after installation, or did this thing manually? You have to add howdy to pam and configure selinux, otherwise howdy won't even be started on login/sudo.
Solution of creating new directory was only to python exception when `Unknown error: 1` was showing.

rtccreator, 2023-08-23:
Same here
`
[ INFO:0@0.625] global registry.impl.hpp:114 UIBackendRegistry UI: Enabled backends(3, sorted by priority): GTK(1000); GTK3(990); GTK2(980) + BUILTIN(QT5)
QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[ INFO:0@0.707] global registry_parallel.impl.hpp:96 ParallelBackendRegistry core(parallel): Enabled backends(2, sorted by priority): TBB(1000); OPENMP(990)
[ INFO:0@0.707] global parallel_for.tbb.hpp:54 ParallelForBackend Initializing TBB parallel backend: TBB_INTERFACE_VERSION=11103
[ INFO:0@0.707] global parallel.cpp:77 createParallelForAPI core(parallel): using backend: TBB (priority=1000)
[ INFO:0@0.710] global cap_v4l.cpp:1973 controlInfo VIDEOIO(V4L2:/dev/video2): property unknown is not supported

`

closed 2026-11-16 by boltgolt