boltgolt/howdy #822: Howdy unlocked my PC using an image

ghaurii, 2023-08-01:
I installed Howdy and did all the config and was working fine.

I tested it out and was fine but I tested it out to see if it unlocks using my image on my mobile.
I took a selfie and went to the login page. Then I hid my face and **showed the picture** on my camera.

Unfortunately, it got unlocked and **is not secure.**

boltgolt, 2023-08-01:
Did you read the bottom of the readme? Are you using an IR or visible light camera? Very difficult to get a phone screen to work when using a proper IR camera

ghaurii, 2023-08-01:
I am using C922 PRO HD STREAM WEBCAM.
Link of the webcam:
https://www.logitech.com/en-us/products/webcams/c922-pro-stream-webcam.960-001087.html

closed 2023-08-01 by boltgolt