boltgolt/howdy #823: Bypass first time configuration during apt install

rosydawn6, 2023-08-04:
Hello.  First thank you for such a great FR login system.  It is very useful.  I am trying to automate the installation via a script and am wondering if there is a way to bypass the interactive prompts for configuration or pass in default values.  Ideally if they could be skipped I could use sed to put in the desired values after installation.

Thanks

adamlove86, 2023-10-22:
I normally do 
echo f | sudo apt install python3 howdy -y

rosydawn6, 2023-10-22:
Learn something new every day.  Great tip thanks!

closed 2023-10-22 by rosydawn6

oxivanisher, 2024-09-18:
To anyone else finding this thread, the ubuntu ppa package file `preinst` checks for the `HOWDY_NO_PROMPT` environment variable. Set this variable to skip the setup.