boltgolt/howdy #825: sudo triggers two successive face scans

badaz, 2023-08-12:
Hello,

whenever I use sudo in the terminal, my face is scanned twice.

There is a first scan (webcam IR scanner blinking) which then displays message "Identified face as {username}" followed by a second scan (webcam IR scanner blinking) which also ends up displaying message "Identified face as {username}". Then the sudo command is executed.

Therefore it does work, but using sudo takes twice the time it should take since my face is scanned twice.

I've tried this :

First test :
- successfull scan the first time
- then I hide my face for the second scan to fail after timeout
- sudo command works as intended

it seems only first scan is required

Second test :
 - first scan fails (hide my face then timeout)
 - second scan is not performed, my password is required instead
 - type my password
 - second scan is triggered
 - hide my face until timeout
 - sudo command works as intended

In any case it seems that this second scan is utterly useless, and I'd love to understand why it is even triggered

NB: login screen only scans my face once and works well, except that once logged in I'm still asked for my password to unlock the keyring, but I guess this is another issue

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu 23.04

Howdy version (`sudo howdy version`): 2.6.1

badaz, 2023-08-13:
Here is the output of `tail -f /var/log/auth.log` when face is scanned successfully twice :

```sh
2023-08-13T16:34:36.109483+02:00 ubuntu [HOWDY]: Attempting facial authentication for user username
2023-08-13T16:34:37.890640+02:00 ubuntu [HOWDY]: Login approved
2023-08-13T16:34:37.890986+02:00 ubuntu [HOWDY]: Attempting facial authentication for user username
2023-08-13T16:34:39.528045+02:00 ubuntu [HOWDY]: Login approved
2023-08-13T16:34:39.528113+02:00 ubuntu sudo:      username : TTY=pts/1 ; PWD=/home/username ; USER=root ; COMMAND=/usr/bin/apt update
2023-08-13T16:34:39.528616+02:00 ubuntu sudo: pam_unix(sudo:session): session opened for user root(uid=0) by (uid=1000)
2023-08-13T16:34:41.842248+02:00 ubuntu sudo: pam_unix(sudo:session): session closed for user root
```

badaz, 2023-09-21:
Am I the only one facing this problem?

badaz, 2023-09-25:
I guess I am then ;)

badaz, 2023-10-03:
I finally found a solution (though I am not sure I understand everything about it). My searches about having to authenticate twice when using sudo lead me to this thread https://ubuntuforums.org/showthread.php?t=817058

In the end the solution was to update the /etc/pam.d/common-auth manually and change this line:
```sh
auth	[success=3 default=ignore]        pam_python.so /lib/security/howdy/pam.py
```

with this:
```sh
auth	sufficient        pam_python.so /lib/security/howdy/pam.py
```

closed 2023-10-03 by badaz

rschrenk, 2024-12-01:
Hi @badaz 
I had exactly the same problem - you are ingenious! Your solution solved it!
Best wishes