boltgolt/howdy #833: howdy Debian12.1 Error

cektor, 2023-08-29:
**Errors I get when I try to install howdy on Debian 12.1 via PPA and .dep.**

```
Unpacking: howdy (2.6.1) ...
Setting: howdy (2.6.1) ...

>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
     python3-xyz, where xyz is the package you are trying to
     install.
    
     If you wish to install a non-Debian-packaged Python package,
     create a virtual environment using python3 -m venv path/to/venv.
     Then use path/to/venv/bin/python and path/to/venv/bin/pip. make
     sure you have python3-full installed.
    
     If you wish to install a non-Debian packaged Python application,
     it may be easiest to use pipx install xyz, which will manage a
     virtual environment for you. Make sure you have pipx installed.
    
     See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: problem processing howdy package (--install):
  howdy package post-installation script installed subprocess terminated with 1 bad exit code
Processing triggers: man-db (2.11.2-2) ...
Errors encountered during processing:
  howdy

```
------------------------------------------

```
conqueror  Lenovo-Pardus  ~  sudo add-apt-repository ppa:boltgolt/howdy
Repository: 'deb https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu/ twentyuc main'
Description:
       +------ /---\ ------+
       | /----\ /- -\ |
       | /- \--/ -\ |
       | /- | |
                  | |
               /--| |-\
          /----- | | -----\
       /-- | | ---\
     --- | | ---
        \--- ----/
            \--- ----/
                \-------- --------/
                  | \----------/ |
                  | _ _ |
                  | / \ / \ |
                  | |
                  | |
       | \ / |
       | \ / |
       | \ / |
       +------ \----- -----/ ------+
                            \---/

                __ _
               | | | | | |
               | |__| | _____ ____| |_ _
               | __ |/ _ \ \ /\ / / _` | | | |
               | | | | (_) \ V V / (_| | |_| |
               |_| |_|\___/ \_/\_/ \__,_|\__, |
                                           __/ |
                                          |___/

Howdy for Ubuntu

Windows Hello style authentication for Ubuntu. Use your built-in IR emitters and camera in combination with face recognition to prove who you are.

https://github.com/boltgolt/howdy
More info: https://launchpad.net/~boltgolt/+archive/ubuntu/howdy
Adding repository.
Press [ENTER] to continue or Ctrl-c to cancel.
Adding deb entry to /etc/apt/sources.list.d/boltgolt-ubuntu-howdy-yirmiuc.list
Adding disabled deb-src entry to /etc/apt/sources.list.d/boltgolt-ubuntu-howdy-yirmiuc.list
Adding key to /etc/apt/trusted.gpg.d/boltgolt-ubuntu-howdy.gpg with fingerprint 2802FD7D4CBF1B050BDE8F3D8CEBC97F3E9E1D58
Same: 1 http://depo.pardus.org.tr/pardus twentyuc InRelease
Same: 2 http://depo.pardus.org.tr/pardus twentyuc-deb InRelease
Same: 3 http://depo.pardus.org.tr/guvenlik twentyuc-deb InRelease
Same: 4 https://dl.google.com/linux/chrome/deb stable InRelease
Ignore:5 https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu twentyuc InRelease
Error: 6 https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu twentyuc Release
   404 Not Found [IP: 185.125.190.52 443]
Reading package lists... Done
E: There is no Release file in the 'https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu twentyuc Release' repository.
N: It is not possible to securely update from such a repository, so the repository is disabled.
N: Detailed information on repository creation and user configuration can be found on the apt-secure(8) guide page.

  fatih  Lenovo-Pardus  ~  sudo apt update
Same: 1 http://depo.pardus.org.tr/pardus twentyuc InRelease
Same: 2 http://depo.pardus.org.tr/pardus twentyuc-deb InRelease
Same: 3 http://depo.pardus.org.tr/guvenlik twentyuc-deb InRelease
Same: 4 https://dl.google.com/linux/chrome/deb stable InRelease
Ignore:5 https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu twentyuc InRelease
Error: 6 https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu twentyuc Release
   404 Not Found [IP: 185.125.190.52 443]
Reading package lists... Done
E: There is no Release file in the 'https://ppa.launchpadcontent.net/boltgolt/howdy/ubuntu twentyuc Release' repository.
N: It is not possible to securely update from such a repository, so the repository is disabled.
N: Detailed information on repository creation and user configuration can be found on the apt-secure(8) guide page.

  fatih  Lenovo-Pardus  ~  sudo apt install howdy
Reading package lists... Done
Creating dependency tree... Done
Reading status info... Done
E: howdy package not found
```

Howdy version (`sudo howdy version`): 2.6.1.

athdromeda, 2023-09-17:
Same. Then I install python3-venv and run ```man howdy``` it works, but ```howdy``` or ```sudo howdy``` give me ```command not found```.

Garret-Martin, 2023-09-17:
I got a jank solution by adding these lines to ~/.config/pip/pip.conf:
[global]
break-system-packages = true
Setting up a virtual environment would be a better option as to not disrupt any dependencies from apt. I had some decencies not automatically installed that I had to do manually. If the howdy commands are working but nothing happens when using sudo, check the /var/log/auth.log. I had a problem with ConfigParser, which I had to rename in the source to be configparser (its a weird pip thing). These are the problems I had, and I hope this helps; but something definitely needs to be fixed with the debian installation.

musikid, 2023-10-13:
Closing this as it is a duplicate of #807, feel free to reopen it if you don't agree.

closed 2023-10-13 by musikid