boltgolt/howdy #836: Beta crashes with no face model known

tokox, 2023-09-13:
While testing beta version I discovered that when there is no face model known for user, then howdy crashes with this error:

```
Child killed by signal User defined signal 1 (10)
```

However, this error is definitely not right. 'pam_howdy' is detecting 'no face model' there (checked it):

https://github.com/boltgolt/howdy/blob/f7649fc922b5b7877146d00d0475bad9d6453531/howdy/src/pam/main.cc#L254-L259

Error steps:
1. howdy_status is called
2. status isn't success, so
3. howdy_error is called
4. status isn't child return value or signal (undefined behaviour probably? (idk how `WIFEXITED` and `WIFSIGNALED` work))
5. second if is executed
6. 'Child killed...' message logged
7. error is returned to pam

Maybe separate function to detect type of CompareError and what happened to child will be a good solution? What do you think?

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Fedora 38

Howdy version (`sudo howdy version`): Howdy 3.0.0 BETA

musikid, 2023-09-13:
#803 contains a fix for this.

tokox, 2023-09-13:
OK, didn't checked there. When will it be merged? Thanks

boltgolt, 2023-09-14:
Just now!

PR #803 (2023-09-14): build: add meson build system 
Based on #773, add Meson as a build system for the whole howdy package,
which allows better flexibility for the paths' configuration.
Generate a python module,
which contains the paths used by the other modules,
by using `PurePath` to simplify path concatenation and limit errors,
and the header containing the paths for the PAM module.

Closes #836
closed 2023-09-14 by boltgolt