boltgolt/howdy #838: New version of python throwing this error

solankybasant, 2023-09-16:
>>> Upgrading pip to the latest version

error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.

    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.

    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.11.2-1) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

Garret-Martin, 2023-09-17:
I added these lines to ~/.config/pip/pip.conf:
[global]
break-system-packages = true

This is a bad solution but it can work. Pip will through a warning every time, because its a bad idea to mix dependencies from different package managers in a non-virtual environment.

musikid, 2023-10-03:
I'm closing this as it is a duplicate of #807, feel free to reopen it if you don't agree.

closed 2023-10-03 by musikid

ShadowStorm0, 2023-11-11:
I tried the solution proposed my @Garret-Martin, however I still get `error: externally-managed-environment` after running `apt install howdy`

`Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit
 status 1
Processing triggers for man-db (2.11.2-1) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)`

dominhnhut01, 2025-01-13:
> I added these lines to ~/.config/pip/pip.conf: [global] break-system-packages = true
> 
> This is a bad solution but it can work. Pip will through a warning every time, because its a bad idea to mix dependencies from different package managers in a non-virtual environment.

For anyone reading this, don't try this if you're on Ubuntu or Linux Mint, it can break the OS

DaanSelen, 2025-02-24:
> For anyone reading this, don't try this if you're on Ubuntu or Linux Mint, it can break the OS

How else can we fix Howdy?