boltgolt/howdy #840: Howdy activated on sleep

cloudmadeofcandy, 2023-09-25:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

As my laptop goes to sleep, Howdy is activated. This poses a security issue, where Howdy would unlock my computer as it is transitioning to sleep mode.
----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Ubuntu Unity 22.04

Howdy version (`sudo howdy version`): Howdy 2.6.1

boltgolt, 2023-09-25:
Are you locking your screen first and then sleeping? What is your lockscreen program?

SimonBrandner, 2024-05-05:
Has there been any investigation into this, feels like a potential security issue...? I found that waiting for the screen to properly lock before letting the device go to sleep might be a functional workaround (discord.com/channels/961691461554950145/1215490910364762143/1236370888396374096)

SimonBrandner, 2024-05-06:
Ok, I've just seen this again with AGS... @boltgolt, could this please receive a little more attention - it's making Howdy a security risk...

closed 2026-11-16 by boltgolt