boltgolt/howdy #841: Howdy opencv module not importing

SpiderUnderUrBed, 2023-10-05:
_Please describe the issue in as much detail as possible, including any errors and traces.

Hello, i am using: https://aur.archlinux.org/packages/howdy-beta-git
The issue i am getting is with the opencv-python module, a dependency of howdy

_If your issue is a camera issue, be sure to also post the image generated by running sudo howdy snapshot
```
[SpiderUnderUrBed@DaSpiderCave ~]$ sudo howdy add
[sudo] password for SpiderUnderUrBed: 
Traceback (most recent call last):
  File "/usr/lib/howdy/cli.py", line 98, in <module>
    import cli.add
  File "/usr/lib/howdy/cli/add.py", line 13, in <module>
    from recorders.video_capture import VideoCapture
  File "/usr/lib/howdy/recorders/video_capture.py", line 6, in <module>
    import cv2
ImportError: libabsl_die_if_null.so.2301.0.0: cannot open shared object file: No such file or directory
```

----

I've searched for similar issues already, and my issue has not been reported yet. ✅

Linux distribution (if applicable): Arch

Howdy version (`sudo howdy version`):
Howdy 3.0.0 BETA

musikid, 2023-12-23:
Did you try to re-install `python-opencv`? It seems to be an issue not on Howdy side.

boltgolt, 2026-11-16:
Hey, thanks for reporting this! This was reported against an install method from before v3. Because v3 introduces a new PAM integration and installation, I think this issue is now stale. Please try the newest version and open a new issue if it's still relevant there!

closed 2026-11-16 by boltgolt