boltgolt/howdy #843: Logging in does not work at all in SDDM or KDE Plasma

176-85-44-91, 2023-10-12:
Neither SDDM nor the plasma lock screen make any attempts at even invoking howdy, I've added `auth sufficient pam_python.so /lib/security/howdy/pam.py` to `/etc/pam.d/system-auth` and `/etc/pam.d/kde`, still no change

----

Found similar issues, but not exactly this one, nor did they have solutions that worked for me

Linux distribution (if applicable):  Fedora 38

Howdy version (`sudo howdy version`): 2.6.1

Garret-Martin, 2023-10-15:
I've been having the same issue. It seems to prefer spamming my camera instead of using the actual password I put in. I had to go back to grub just to disable it.

adamlove86, 2023-10-18:
My install process was generally:

sudo add-apt-repository ppa:boltgolt/howdy -y
sudo apt update
echo f | sudo apt install python3 python-is-python3 howdy -y
#sudo cp /home/adam/Documents/install_stuff/howdy_config.ini /lib/security/howdy/config.ini
sudo chmod -R 755 /lib/security/howdy

I think the chmod helped?

mbronikowski, 2023-12-04:
I'm trying to set this up on Kubuntu 23.10 and I'm facing the same issue. After doing the chmod, howdy gets my login screen stuck on an infinite loop of "Authentication failed", from which the only escape seems to be a reboot. Everything works correctly with sudo howdy test - the correct IR camera is invoked etc.

a15355447898a, 2023-12-15:
The same problem, I am using archlinux, sddm+plasma, and howdy-beta-git from AUR.

I modified the `sudo`, `kde`, `sddm`, `login`, `system-local-login`, and `polkit-1` files under the `/etc/pam.d` directory.

On the sddm interface, howdy is not called directly for unlocking. It is only called after I enter the password and press enter, regardless of whether the password is correct or not.

If the previous step's password is incorrect, I can enter the plasma desktop, but the plasma desktop keeps restarting. After a few restarts, the desktop becomes all black, with only my mouse still visible.

If the previous step's password is correct, I can enter the plasma desktop normally and use it as usual.

My plasma lock screen can directly call howdy and unlock and use normally.

closed 2026-11-16 by boltgolt