boltgolt/howdy #845: Package not found (Ubuntu 23.10)

dm30, 2023-10-22:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

Hi! This is happening with many programs.

javi@Ubuntu:~$ sudo apt install howdy
Leyendo lista de paquetes... Hecho
Creando árbol de dependencias... Hecho
Leyendo la información de estado... Hecho
E: No se ha podido localizar el paquete howdy // Excuse the spanish: Package could not be found

How can it possibly be? Fault on Canonical side?
Thanks in advance.

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):

Howdy version (`sudo howdy version`):

javier-nogales, 2023-10-23:
Hi dm30! The problem is there is not version for ubuntu 23.10 (mantic) and the package doesn't exists for 23.10. 
When you add ppa repository like this: `add-apt-repository ppa:boltgolt/howdy,` your system write this `add-apt-repository ppa:boltgolt/howdy mantic main`. You can wach it in "Sofware & Upadtes" >" Other Software". 
You can try to change version manualy to "jammy" but it doesn't work either.

dm30, 2023-10-23:
Thank you, isn't that regarding to updates. The howdy package itself is not available in Ubuntu 23.10 repositories. Anyway, should not throw an error when the package / PPA was not updated for current Ubuntu version?

closed 2023-10-23 by boltgolt