boltgolt/howdy #847: Adding pipx support

dm30, 2023-10-23:
Many months since, installation fails even in new systems because of dkpg error 1 linked to this python - pipx thing.
Searching for "error 1" in Issues will return some examples. 
Great if could be fixed. I really miss face recognition logging. 
Thank you. 

Se encontraron errores al procesar:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)

ShadowStorm0, 2023-11-10:
I second this, please add pipx support

dm30, 2023-11-10:
I had to resort to Ubuntu 22.04 LTS. I hope something can be done for Noble Numbat release!

musikid, 2023-11-14:
Since this is a packaging issue related to #807, I will close this issue, so we can track this in a centralized issue.

closed 2023-11-14 by musikid