boltgolt/howdy #849: Unknown error: 1

DocCringol, 2023-10-30:
So i can't make howdy work

`howdy test` is working, `sudo -i` isn't working and returning
```
sudo howdy version
Unknown error: 1
```

What i've done:
- install from fusion
- add `/dev/video2` to `sudo howdy config`
- add `auth sufficient pam_python.so /lib64/security/howdy/pam.py` to `/etc/pam.d/sudo`
- `chmod o-x /lib64/security/howdy/compare.py`

Also tried (maybe the wrong way):
- https://github.com/boltgolt/howdy/issues/233
- https://github.com/boltgolt/howdy/issues/830
- https://discussion.fedoraproject.org/t/anbinx-howdy/46321

Logs:
```
окт 30 10:20:00 MSI [8580]: Attempting facial authentication for user doccringol
окт 30 10:20:05 MSI python3[8582]: detected unhandled Python exception in '/lib64/security/howdy/compare.py'
окт 30 10:20:05 MSI [8580]: Failure, unknown error1
окт 30 10:20:05 MSI audit: BPF prog-id=100 op=LOAD
окт 30 10:20:05 MSI systemd[1]: Starting fprintd.service - Fingerprint Authentication Daemon...
окт 30 10:20:05 MSI abrt-server[8633]: Package 'howdy' isn't signed with proper key
окт 30 10:20:05 MSI abrt-server[8633]: 'post-create' on '/var/spool/abrt/Python3-2023-10-30-10:20:05-8582' exited with 1
окт 30 10:20:05 MSI abrt-server[8633]: Deleting problem directory '/var/spool/abrt/Python3-2023-10-30-10:20:05-8582'
окт 30 10:20:05 MSI systemd[1]: Started fprintd.service - Fingerprint Authentication Daemon.
окт 30 10:20:05 MSI audit[1]: SERVICE_START pid=1 uid=0 auid=4294967295 ses=4294967295 subj=system_u:system_r:init_t:s0 msg='unit=fprintd comm="systemd" exe="/usr/lib/systemd/systemd" hostname=? addr=? terminal=? res=success'
```

Howdy version: `2.6.1`

tokox, 2023-10-30:
have you configured SELinux and #801?

> chmod o-x /lib64/security/howdy/compare.py

it should be `o+x` not `o-x`

DocCringol, 2023-10-30:
> it should be `o+x` not `o-x`
yeah, missclick

and yes, #801 helped, thanks a lot

closed 2023-10-30 by DocCringol