boltgolt/howdy #851: Missing cli module in ubuntu 23.10

g1ronn1mo, 2023-11-02:
I tryed to add a new user by
`$ sudo howdy  add `

Error:
```
Traceback (most recent call last):
  File "/usr/local/bin/howdy", line 95, in <module>
    import cli.add
ModuleNotFoundError: No module named 'cli'

```

Linux distribution (if applicable): Ubuntu 23.10

Howdy version (`sudo howdy version`): Howdy 2.6.1

I think that has something to do how the new ubuntu handels its python environment. I tried to install the cli package with pipx, but it did not find the module.

JasonW41k3r, 2023-11-05:
Same problem as you, ubuntu 23.10

2023323-MarcosOliveira, 2023-11-20:
Same here guys, Ubuntu 23.10, GNOME 45.

1. I tried:
```
sudo add-apt-repository ppa:boltgolt/howdy
sudo apt update
sudo apt install howdy
```
2. Howdy found the IR emitters:
```
Starting certainty auto config...

After detection, Howdy knows how certain it is that the match is correct.
How certain Howdy needs to be before authenticating you can be customized.

F: Fast.
Allows more fuzzy matches, but speeds up the scanning process greatly.

B: Balanced.
Still relatively quick detection, but might not log you in when further away.

S: Secure.
The safest option, but will take much longer to authenticate you.

You can always change this setting in the config.
What profile would you like to use? [f/b/s]: f

Unpacking howdy (2.6.1) ...
Setting up howdy (2.6.1) ...

>>> Upgrading pip to the latest version
```

4. Then I got an error:
```
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your
 Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.11.2-3) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

5. Trying to config Howdy
```
marcos@oliveira:~$ sudo howdy config
```
`sudo: howdy: command not found`

6. But Howdy is installed ?!
```
marcos@marcos-oliveira:~$ whereis howdy
```
howdy: /usr/share/man/man1/howdy.1.gz

I am very new to Linux, and am I doing something wrong? I would like to have the same experience when I unlock my Windows, but now, using Linux, which is much much better then Windows.

g1ronn1mo, 2023-12-10:
Got it running with ubuntu 23.10 without breakling any packages. 
0. I dont know if this is necessary, but I did it. 
```
sudo pip install -t /lib/security/howdy dlib   
sudo add-apt-repository ppa:boltgolt/howdy             
sudo apt update
sudo apt install howdy
sudo ln /lib/security/howdy/cli.py /usr/local/bin/howdy
```

Maybe it is easier to follow the instructions on https://github.com/boltgolt/howdy/pull/814. for the installation if the dlib  library. 

1. download the deb12 packages from: https://github.com/boltgolt/howdy/pull/814
2. install them with right click 
3. `sudo howdy config` and change the line starting with device path to: `device_path = /dev/video0`
3. `sudo howdy add`

Let me know if this is working for you.

KeysBit, 2024-01-20:
Not compeletly work. The command ' sudo ln /lib/security/howdy/cli.py /usr/local/bin/howdy 'change the relative diretory of 'cli.py' so the 'import' command in python cannot detect the true module path so I only can run howdy by 'sudo /lib/security/howdy/cli.py'... That`s really stupid, or maybe I cannot use python well...

KeysBit, 2024-01-20:
> Not compeletly work. The command `sudo ln /lib/security/howdy/cli.py /usr/local/bin/howdy` change the relative diretory of `cli.py` so the `import` in python cannot detect the true module path so I only can run howdy by `sudo /lib/security/howdy/cli.py`... That`s really stupid, or maybe I cannot use python well...

Update: Trying to add `sys.path.append('/lib/security/howdy')` in `cli.py` got everything worked.

jarda15, 2024-04-22:
For me worked  sudo /usr/local/bin/python3 -m pip install numpy opencv-python if still not working try ls /usr/local/bin/ and see which python u have installed and then repeat with other version. Like this sudo /usr/local/bin/put_your_python -m pip install numpy opencv-python

einsteinarbert, 2024-09-13:
Same as error:
> https://github.com/boltgolt/howdy/issues/927#issuecomment-2348636233

Kirya02, 2025-06-05:
After installation got an error `howdy command not found`. It worked to run manually `sudo cli.py`. (you should be inside the `/lib/security/howdy/` directory). After that, the command `cli.py` add should work. But I found out that nothing happens if you try to unlock via howdy.
So I tried to run `journalctl -xe | grep howdy`.  found out the file `pam.py`  can not include a `ConfigParser` library because its a python2 library and `configparser` is a python3 library.
So just change `import ConfigParser` to `import configparser as ConfigParser` and it worked for me! (You can ignore the errors it shows about pip when your installing with apt)

closed 2026-11-16 by boltgolt