boltgolt/howdy #856: Sudo does not use howdy on Ubuntu 23.04

emma-freeman-plym, 2023-11-10:
I've added a model, and checked it works with `sudo /lib/security/howdy/cli.py test`. The file `/usr/share/pam-configs/howdy` is correct. `pam-auth-update` lists howdy as enabled.
 But `sudo -i` doesn't enable the IR camera, it just shows password auth. Any idea why?

musikid, 2023-11-13:
What is the content of the `/etc/pam.d/sudo` file?

emma-freeman-plym, 2023-11-14:
Nothing mentioning howdy:
```
#%PAM-1.0

# Set up user limits from /etc/security/limits.conf.
session    required   pam_limits.so

session    required   pam_env.so readenv=1 user_readenv=0
session    required   pam_env.so readenv=1 envfile=/etc/default/locale user_readenv=0

@include common-auth
@include common-account
@include common-session-noninteractive
```

techonerd, 2023-11-19:
> What is the content of the `/etc/pam.d/sudo` file?

```
#%PAM-1.0

# Set up user limits from /etc/security/limits.conf.
auth   sufficient        pam_python.so /lib/security/howdy/pam.py
session    required   pam_limits.so

@include common-auth
@include common-account
@include common-session-noninteractive
```

Is this the expected output? I think I added `**/howdy/pam.py` there manually by following some troubleshooting guides. Not sure if this is the expected content of the file.

musikid, 2023-12-26:
This isn't the expected output indeed, it should have an entry for howdy. If you put this in the file, it should work.
```
auth   sufficient        pam_python.so /lib/security/howdy/pam.py
```

closed 2026-11-16 by boltgolt