boltgolt/howdy #858: Ubuntu 23.10, GNOME 45

2023323-MarcosOliveira, 2023-11-20:
I'm very new to Linux and I would like to install Howdy, to get the same experience I had on Windows.

## System Details Report

```
**Date generated:**                              2023-11-20 20:07:20

## Hardware Information:
**Hardware Model:**                              Razer Blade 15 Advanced Model _Mid 2021_ - RZ09-0409
**Memory:**                                      32.0 GiB
**Processor:**                                   11th Gen Intel® Core™ i7-11800H × 16
**Graphics:**                                    Intel® UHD Graphics (TGL GT1)
**Graphics 1:**                                  NVIDIA GeForce RTX™ 3070 Laptop GPU
**Disk Capacity:**                               1.5 TB

## Software Information:
**Firmware Version:**                            2.02
**OS Name:**                                     Ubuntu 23.10
**OS Build:**                                    (null)
**OS Type:**                                     64-bit
**GNOME Version:**                               45.0
**Windowing System:**                            Wayland
**Kernel-Version:**                              Linux 6.5.0-10-generic
```

## Steps

1. I tried:
```
sudo add-apt-repository ppa:boltgolt/howdy
sudo apt update
sudo apt install howdy
```
2. Howdy found the IR emitters:
```
Starting certainty auto config...

After detection, Howdy knows how certain it is that the match is correct.
How certain Howdy needs to be before authenticating you can be customized.

F: Fast.
Allows more fuzzy matches, but speeds up the scanning process greatly.

B: Balanced.
Still relatively quick detection, but might not log you in when further away.

S: Secure.
The safest option, but will take much longer to authenticate you.

You can always change this setting in the config.
What profile would you like to use? [f/b/s]: f

Unpacking howdy (2.6.1) ...
Setting up howdy (2.6.1) ...

>>> Upgrading pip to the latest version
```

4. Then I got an error:
```
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your
 Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
Error while running last command
dpkg: error processing package howdy (--configure):
 installed howdy package post-installation script subprocess returned error exit status 1
Processing triggers for man-db (2.11.2-3) ...
Errors were encountered while processing:
 howdy
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

5. Trying to config Howdy
```
marcos@oliveira:~$ sudo howdy config
```
`sudo: howdy: command not found`

6. But Howdy is installed ?!
```
marcos@marcos-oliveira:~$ whereis howdy
```
howdy: /usr/share/man/man1/howdy.1.gz

## Conclusion:
I do not know what to do, can you guys help me, please? Thanks.

esa1975, 2023-11-21:
Hello. I am using Ubuntu 23.10, have added the PPA, and received the same error message when installing. It appears as though all the dependencies install properly but the pip portion fails. I don't know how to take it any further so I am reporting it here in hopes that you someone can assist. Also, thank you for the excellent project. I have used howdy on other systems and find it very useful.

jamopopper, 2023-11-26:
I'm also getting the same error. Howdy was one of the main reasons I was switching to Linux (aside from gaming getting better, computer science major, etc.). Hopefully this gets fixed soon

Ankan-cyber, 2023-11-28:
yeah same issue but I am using gnome 45 with debain 12 I couldn't installed using ppa so I used the direct deb files first it gave me error then I installed the dependency then it installed the first install script runs but the pip external error popping up I installed it in a separate ubuntu vm its works just fine if anyone has some fix reply

Sciroccogti, 2023-11-30:
#781 This issue may help.

g1ronn1mo, 2023-12-10:
Found the solution:

https://github.com/DingoBits/howdy/releases/tag/3.0.0-git20230625.c17a834

1. download the deb 12 packages:

  [howdy_3.0.0.git20230625.c17a834-1_amd64_deb12.deb](https://github.com/dingobits/howdy/releases/download/3.0.0-git20230625.c17a834/howdy_3.0.0.git20230625.c17a834-1_amd64_deb12.deb)
  and
  [python3-dlib_19.24.2-1_amd64_deb12.deb](https://github.com/dingobits/howdy/releases/download/3.0.0-git20230625.c17a834/python3-dlib_19.24.2-1_amd64_deb12.deb)

2. Right click and click: "open with software software install" 

3. run "sudo howdy add" will resuilt in:

  ```
  Traceback (most recent call last):
    File "/lib/security/howdy/compare.py", line 19, in <module>
      import cv2
  ModuleNotFoundError: No module named 'cv2'
  nknown error: 
  [sudo] password for benedikt-schwab: 
  Data files have not been downloaded, please run the following commands:
  
	  cd /etc/howdy/dlib-data
	  sudo ./install.sh
  ```

4. run 

  ```
  cd /etc/howdy/dlib-data
  sudo ./install.sh
  ```
  
  5.  run "sudo howdy add"  agin will result in:
  
```
Adding face model for the user benedikt-schwab
Enter a label for this new model [Initial model]: bene
Howdy could not find a camera device at the path specified in the config file.
It is very likely that the path is not configured correctly, please edit the 'device_path' config value by running:

	sudo howdy config

```
  
So you must input your camera in the config.  This sould be described somewhere. I will give an update.

Ankan-cyber, 2023-12-10:
yeah its works @g1ronn1mo thanks for the fix

jamopopper, 2023-12-12:
> yeah its works @g1ronn1mo thanks for the fix

can confirm, I have also gotten it running now

7HE-W0R1D, 2024-01-02:
> Found the solution:
> 
> https://github.com/DingoBits/howdy/releases/tag/3.0.0-git20230625.c17a834
> 
> 1. download the deb 12 packages:
> 
> [howdy_3.0.0.git20230625.c17a834-1_amd64_deb12.deb](https://github.com/dingobits/howdy/releases/download/3.0.0-git20230625.c17a834/howdy_3.0.0.git20230625.c17a834-1_amd64_deb12.deb) and [python3-dlib_19.24.2-1_amd64_deb12.deb](https://github.com/dingobits/howdy/releases/download/3.0.0-git20230625.c17a834/python3-dlib_19.24.2-1_amd64_deb12.deb)
> 
> 2. Right click and click: "open with software software install"
> 3. run "sudo howdy add" will resuilt in:
> 
> ```
> Traceback (most recent call last):
>   File "/lib/security/howdy/compare.py", line 19, in <module>
>     import cv2
> ModuleNotFoundError: No module named 'cv2'
> nknown error: 
> [sudo] password for benedikt-schwab: 
> Data files have not been downloaded, please run the following commands:
> 
>     cd /etc/howdy/dlib-data
>     sudo ./install.sh
> ```
> 
> 4. run
> 
> ```
> cd /etc/howdy/dlib-data
> sudo ./install.sh
> ```
> 
> 5. run "sudo howdy add"  agin will result in:
> 
> ```
> Adding face model for the user benedikt-schwab
> Enter a label for this new model [Initial model]: bene
> Howdy could not find a camera device at the path specified in the config file.
> It is very likely that the path is not configured correctly, please edit the 'device_path' config value by running:
> 
> 	sudo howdy config
> ```
> 
> So you must input your camera in the config. This sould be described somewhere.
> 
> In the end I changed the camera settings to:
> 
> `device_path = /dev/video0`
> 
> Worked for me at least.

Thanks for the fix @g1ronn1mo 

Also a note if you are using IR camera, then check your available video devices with `ls -ltrh /dev/video*`. I have four,  tested them one by one in `howdy config` and `/dev/video2` is my IR camera.

2023323-MarcosOliveira, 2024-01-04:
> Found the solution:
>
> https://github.com/DingoBits/howdy/releases/tag/3.0.0-git20230625.c17a834
>
> 1. download the deb 12 packages: [...]

I also can confirm it works 100%. I wish it was implemented by default in GNOME, in the welcome screen when first installed Ubuntu or any other Linux distribution. Thanks guys!

closed 2024-01-04 by 2023323-MarcosOliveira