boltgolt/howdy #860: Segmentation fault with fresh install on fresh openSUSE Tumbleweed, with x11 and KDE Plasma

Ultome, 2023-12-01:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

I just installed howdy and I wanted to check it worked... But it didn't. Here's the output of `sudo howdy test` for "/dev/video1":
```
[ WARN:0@0.082] OpenCV | GStreamer warning: Embedded video playback halted; module source reported: Could not read from resource.
[ WARN:0@0.082] OpenCV | GStreamer warning: unable to start pipeline
[ WARN:0@0.082] OpenCV | GStreamer warning: GStreamer: pipeline have not been created

Opening a window with a test feed

Press ctrl+C in this terminal to quit
Click on the image to enable or disable slow mode

QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-root'
[1]    20729 segmentation fault  sudo howdy test
```
And here the output for `/dev/video0`:
```
(python3:21095): GStreamer-CRITICAL **: 09:55:38.741: 
Trying to dispose element pipeline0, but it is in READY instead of the NULL state.
You need to explicitly set elements to the NULL state before
dropping the final reference, to allow them to clean up.
This problem may also be caused by a refcounting bug in the
application or some element.

[ WARN:0@0.077] OpenCV | GStreamer warning: unable to start pipeline

(python3:21095): GStreamer-CRITICAL **: 09:55:38.741: 
Trying to dispose element videoconvert0, but it is in PAUSED instead of the NULL state.
You need to explicitly set elements to the NULL state before
dropping the final reference, to allow them to clean up.
This problem may also be caused by a refcounting bug in the
application or some element.

[ WARN:0@0.077] OpenCV | GStreamer warning: GStreamer: pipeline have not been created
[1]    21093 segmentation fault  sudo howdy test
```

I don't know what to do...

I searched the issues on this GitHub but didn't find any similar to mine enough to help.

Linux distribution (if applicable): openSUSE Tumbleweed

Howdy version (`sudo howdy version`): Howdy 2.6.1

poltpolt, 2025-01-29:
I  confirm the same in Opensuse SlowRoll and same howdy version

poltpolt, 2025-02-05:
I was able to find a workaround
instead of
`sudo howdy test`
in kde plasma use
`kdesu howdy test`

closed 2026-11-16 by boltgolt