boltgolt/howdy #863: Signature not found in user keyring / First boot login loop / Encrypted home

InsanityFlea, 2023-12-06:
Fresh boot, it detects, gives me the login but then loops back to the login screen.
Using ctrl,alt,f1 (tty1) it detects and logs me in. 
It throws up the following issue which I believe is why the first boot login doesn't work.
Home directory is encrypted - not sure if this is supported.

Signature not found in user keyring
Perhaps try run the interactive 'encrypt-mount-private'
To run a command as administrator (user "root"), use "sudo <command>".
See "man sudo_root" for details.

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): Linux Mint 21.2 Cinnamon

Howdy version (`sudo howdy version`): Howdy 2.6.1

tokox, 2023-12-10:
I think this is the same case as #438. Looks like signature for your home encryption is saved in keyring - which is encrypted using your password.

>  Using ctrl,alt,f1 (tty1) it detects and logs me in.

It logs you in - but if I'm not wrong, you can't access your files without providing password? (after boot)

InsanityFlea, 2023-12-11:
Howdy cannot unlock the keyring at the moment, makes sense, thank you.

closed 2023-12-11 by InsanityFlea