boltgolt/howdy #865: Installation Issue on Ubuntu 23.04 & 23.10.1

4mritGiri, 2023-12-11:
## Howdy Installation Issue on Ubuntu 23.10.1

### Problem:
> Purge and reinstall seems to have run into errors:
> 
> ```
> Setting up howdy (2.6.1) ...
> 
> >>> Upgrading pip to the latest version
> 
> error: externally-managed-environment
> 
> × This environment is externally managed
> ╰─> To install Python packages system-wide, try apt install
>     python3-xyz, where xyz is the package you are trying to
>     install.
>     
>     If you wish to install a non-Debian-packaged Python package,
>     create a virtual environment using python3 -m venv path/to/venv.
>     Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
>     sure you have python3-full installed.
>     
>     If you wish to install a non-Debian packaged Python application,
>     it may be easiest to use pipx install xyz, which will manage a
>     virtual environment for you. Make sure you have pipx installed.
>     
>     See /usr/share/doc/python3.11/README.venv for more information.
> 
> note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
> hint: See PEP 668 for the detailed specification.
> Error while running last command
> dpkg: error processing package howdy (--configure):
>  installed howdy package post-installation script subprocess returned error exit status 1
> Processing triggers for man-db (2.11.2-1) ...
> Processing triggers for libc-bin (2.37-0ubuntu2) ...
> Errors were encountered while processing:
>  howdy
> E: Sub-process /usr/bin/dpkg returned an error code (1)
> ```

I encountered an issue with Howdy installation on Ubuntu version `23.10.1`. The provided solutions seemed to have complications.

### Workaround:
I found a workaround that worked for me, even on a fresh install of Ubuntu 23.10.1. Skip the "apt remove howdy" command to avoid additional errors during sudo authentication.

1. Open a root shell:
   ```sh
   sudo -i

2. Move the EXTERNALLY-MANAGED directory:
   ```sh 
     mv /usr/lib/python3.11/EXTERNALLY-MANAGED /usr/lib/python3.11/DISABLED-EXTERNALLY-MANAGED
  
3. Remove Howdy (skip if fresh install):
   ```sh
    apt remove howdy

4. Install dlib using pip3:
     `pip3 install dlib`

   If pip3 doesn't work, try `pip install dlib`. If that also fails, attempt:
      `sudo pip install --break-system-packages dlib`

5. Reinstall Howdy:
   ```sh
    apt install howdy

8. Edit pam.py to fix import error:
   ```sh
    nano /lib/security/howdy/pam.py

 Change `import ConfigParser` to `import configparser as ConfigParser` you can find on lines 2 to 11.

9. Restore EXTERNALLY-MANAGED directory:
   ```sh
    mv /usr/lib/python3.11/DISABLED-EXTERNALLY-MANAGED /usr/lib/python3.11/EXTERNALLY-MANAGED 

This workaround should help resolve the installation issues in my case. 

While adding face some time you will get device_path problem but i am solve it like this.
   ```sh
    # Should be set automatically by an installer if your distro has one
    # device_path = /dev/v4l/by-path/pci-0000:00:14.0-usb-0:6:1.2-video-index1 # default in my config comment it add new below like this
    device_path = /dev/video0

Ankan-cyber, 2023-12-11:
Try this fix it works for me
https://github.com/boltgolt/howdy/issues/858#issuecomment-1848893244

4mritGiri, 2023-12-12:
### Solution:
I found a workaround that worked for me, even on a fresh install of Ubuntu 23.10.1. Skip the "apt remove howdy" command to avoid additional errors during sudo authentication.

1. Open a root shell:
   ```sh
   sudo -i

2. Move the EXTERNALLY-MANAGED directory:
   ```sh 
     mv /usr/lib/python3.11/EXTERNALLY-MANAGED /usr/lib/python3.11/DISABLED-EXTERNALLY-MANAGED
  
3. Remove Howdy (skip if fresh install):
   ```sh
    apt remove howdy

4. Install dlib using pip3:
     `pip3 install dlib`

   If pip3 doesn't work, try `pip install dlib`. If that also fails, attempt:
      `sudo pip install --break-system-packages dlib`

5. Reinstall Howdy:
   ```sh
    apt install howdy

8. Edit pam.py to fix import error:
   ```sh
    nano /lib/security/howdy/pam.py

 Change `import ConfigParser` to `import configparser as ConfigParser` you can find on lines 2 to 11.

9. Restore EXTERNALLY-MANAGED directory:
   ```sh
    mv /usr/lib/python3.11/DISABLED-EXTERNALLY-MANAGED /usr/lib/python3.11/EXTERNALLY-MANAGED 

This workaround should help resolve the installation issues in my case. 

While adding face some time you will get device_path problem but i am solve it like this.
   ```sh
    # Should be set automatically by an installer if your distro has one
    # device_path = /dev/v4l/by-path/pci-0000:00:14.0-usb-0:6:1.2-video-index1 # default in my config comment it add new below like this
    device_path = /dev/video0

nabheet, 2024-02-14:
here is an alternate fix:

```
$ sudo apt-get update 
$ sudo apt-get install howdy; sudo apt-get install howdy pipenv;
$ cd /lib/security/howdy
$ sudo pipenv install numpy opencv-python dlib
$ cat howdy
#!/bin/bash
export PIPENV_PIPFILE=/lib/security/howdy/Pipfile
/usr/bin/pipenv run python /lib/security/howdy/cli.py $@
$ chmod a+x howdy
$ ll howdy
-rwxr-xr-x 1 root root 99 Feb 13 19:36 howdy*
$ sudo ln -s /lib/security/howdy/howdy /usr/local/bin/howdy
$ sudo howdy config
$ sudo howdy add
```

joezhouchenye, 2024-03-12:
> ### Solution:
> I found a workaround that worked for me, even on a fresh install of Ubuntu 23.10.1. Skip the "apt remove howdy" command to avoid additional errors during sudo authentication.
> 
> 1. Open a root shell:
>    ```shell
>    sudo -i
>    ```
> 2. Move the EXTERNALLY-MANAGED directory:
>    ```shell
>      mv /usr/lib/python3.11/EXTERNALLY-MANAGED /usr/lib/python3.11/DISABLED-EXTERNALLY-MANAGED
>    ```
> 3. Remove Howdy (skip if fresh install):
>    ```shell
>     apt remove howdy
>    ```
> 4. Install dlib using pip3:
>    `pip3 install dlib`
>    If pip3 doesn't work, try `pip install dlib`. If that also fails, attempt:
>    `sudo pip install --break-system-packages dlib`
> 5. Reinstall Howdy:
>    ```shell
>     apt install howdy
>    ```
> 6. Edit pam.py to fix import error:
>    ```shell
>     nano /lib/security/howdy/pam.py
>    ```
> 
> Change `import ConfigParser` to `import configparser as ConfigParser` you can find on lines 2 to 11.
> 
> 9. Restore EXTERNALLY-MANAGED directory:
>    ```shell
>     mv /usr/lib/python3.11/DISABLED-EXTERNALLY-MANAGED /usr/lib/python3.11/EXTERNALLY-MANAGED 
>    ```
> 
> This workaround should help resolve the installation issues in my case.
> 
> While adding face some time you will get device_path problem but i am solve it like this.
> 
> ```shell
>  # Should be set automatically by an installer if your distro has one
>  # device_path = /dev/v4l/by-path/pci-0000:00:14.0-usb-0:6:1.2-video-index1 # default in my config comment it add new below like this
>  device_path = /dev/video0 
> ```

The last step is not needed for my case. During the apt install, it will ask if your IR emitter is on. Choose no if your camera light isn't on. It will try to automatically find your camera.

musikid, 2024-04-15:
I copied @4mritGiri's solution (thanks!) in the wiki so people can fix it until the new version.

closed 2024-04-15 by musikid

LeonardoMantovani, 2024-07-06:
I'm having the same issue with Ubuntu 24.04, but neither of the proposed fixes works out...

With @4mritGiri one I get an error when it tries to update pip (`ERROR: Cannot uninstall pip 24.0, RECORD file not found. Hint: The package was installed by debian.`)

With @nabheet solution it fails when I try to add a new model with `sudo howdy add` saying  that I'm using an unsupported image type (`RuntimeError: Unsupported image type, must be 8bit gray or RGB image.`)

What should I do? Where can I find documentation on how to bayass this issue (I've seen a ton of issues regarding this problem but all of them with different workarounds, and apparently none of them really working for me)?

Also, since Ubuntu 24.04 is an LTS release (that will probably become the base of many other distros in the upcoming months), I would expect Howdy to work out-of-the-box on it.
I undestrand that the changes they made with how the system python is managed are breaking changes, but maybe howdy should adapt and use something like `conda` or `pipx` instead of the good old `pip`?

clorteau, 2024-07-11:
Same as @LeonardoMantovani exactly on Kubuntu 24.04. Tried both proposed solutions with the same outcomes.

yangtaihong59, 2024-08-08:
Same as @LeonardoMantovani  exactly on Ubuntu 24.04.

jkatnik, 2024-08-27:
@LeonardoMantovani 
> With @nabheet solution it fails when I try to add a new model with `sudo howdy add` saying that I'm using an unsupported image type (`RuntimeError: Unsupported image type, must be 8bit 
gray or RGB image.`)

Probably, you have two (or more) cameras, the regular one and IR. You have incorrectly selected the first one.