boltgolt/howdy #866: ModuleNotFoundError: No module named 'cv2'

g1ronn1mo, 2023-12-12:
I got this error:

```
Traceback (most recent call last):
  File "/lib/security/howdy/compare.py", line 19, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'

```

freshgum-bubbles, 2023-12-16:
Are you using that "recommended" package on openSUSE? It's *completely* broken.  For some reason the package "depends" upon OpenCV3, where the runtime *actually* uses OpenCV2.

I'm not familiar with OpenCV (or Python) in general, but I've read that v3 can be substantially faster than v2; therefore, I'm wondering if any efforts are being made to migrate the project over to it.

musikid, 2024-04-15:
Feel free to re-open it if you can provide more information.

closed 2024-04-15 by musikid

lindonm, 2024-08-05:
For what it's worth - I'm no linux or python expert, but after trying lots of stuff online I ended up with:

Run sudo python
Run each of the import lines from cli/test.py - see which fails. eg:
```
import configparser
import os
import sys
import time
import dlib
import cv2
```
Had to install dlib using pip install dlib - worked fine
Then tried getting cv2 (opencv-python) a bunch of ways, but the only way that worked for me was 
```
sudo apt install -y python3-opencv
```

Test now running and trying to get the rest set up.