boltgolt/howdy #870: Howdy install error

ab-kaium, 2023-12-17:
_Please describe the issue in as much detail as possible, including any errors and traces._
_If your issue is a camera issue, be sure to also post the image generated by running `sudo howdy snapshot`._

98%] Building CXX object CMakeFiles/_dlib_pybind11.dir/src/gui.cpp.o
[100%] Linking CXX shared module /home/ab-kaium/.cache/yay/python-dlib/src/dlib-19.24.2-cuda/build/lib.linux-x86_64-cpython-311/_dlib_pybind11.cpython-311-x86_64-linux-gnu.so
lto1: fatal error: bytecode stream in file ‘dlib_build/libdlib.a’ generated with LTO version 12.0 instead of the expected 13.0
compilation terminated.
lto-wrapper: fatal error: /usr/bin/c++ returned 1 exit status
compilation terminated.
/usr/bin/ld: error: lto-wrapper failed
collect2: error: ld returned 1 exit status
make[2]: *** [CMakeFiles/_dlib_pybind11.dir/build.make:522: /home/ab-kaium/.cache/yay/python-dlib/src/dlib-19.24.2-cuda/build/lib.linux-x86_64-cpython-311/_dlib_pybind11.cpython-311-x86_64-linux-gnu.so] Error 1
make[1]: *** [CMakeFiles/Makefile2:118: CMakeFiles/_dlib_pybind11.dir/all] Error 2
make: *** [Makefile:91: all] Error 2
Traceback (most recent call last):
  File "/home/ab-kaium/.cache/yay/python-dlib/src/dlib-19.24.2-cuda/setup.py", line 222, in <module>
    setup(
  File "/usr/lib/python3.11/site-packages/setuptools/__init__.py", line 103, in setup
    return distutils.core.setup(**attrs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 185, in setup
    return run_commands(dist)
           ^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/site-packages/setuptools/_distutils/core.py", line 201, in run_commands
    dist.run_commands()
  File "/usr/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 969, in run_commands
    self.run_command(cmd)
  File "/usr/lib/python3.11/site-packages/setuptools/dist.py", line 963, in run_command
    super().run_command(command)
  File "/usr/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/usr/lib/python3.11/site-packages/setuptools/_distutils/command/build.py", line 131, in run
    self.run_command(cmd_name)
  File "/usr/lib/python3.11/site-packages/setuptools/_distutils/cmd.py", line 318, in run_command
    self.distribution.run_command(command)
  File "/usr/lib/python3.11/site-packages/setuptools/dist.py", line 963, in run_command
    super().run_command(command)
  File "/usr/lib/python3.11/site-packages/setuptools/_distutils/dist.py", line 988, in run_command
    cmd_obj.run()
  File "/home/ab-kaium/.cache/yay/python-dlib/src/dlib-19.24.2-cuda/setup.py", line 134, in run
    self.build_extension(ext)
  File "/home/ab-kaium/.cache/yay/python-dlib/src/dlib-19.24.2-cuda/setup.py", line 174, in build_extension
    subprocess.check_call(cmake_build, cwd=build_folder)
  File "/usr/lib/python3.11/subprocess.py", line 413, in check_call
    raise CalledProcessError(retcode, cmd)
subprocess.CalledProcessError: Command '['cmake', '--build', '.', '--config', 'Release', '--', '-j4']' returned non-zero exit status 2.
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: python-dlib-exit status 4
==> Making package: python-face_recognition 1.3.0-1 (Sun 17 Dec 2023 01:38:45 PM +06)
==> Checking runtime dependencies...
==> Missing dependencies:
  -> python-dlib
==> Checking buildtime dependencies...
==> ERROR: Could not resolve all dependencies.
 -> error making: python-face_recognition-exit status 8
==> Making package: howdy 2.6.1-2 (Sun 17 Dec 2023 01:38:46 PM +06)
==> Checking runtime dependencies...
==> Missing dependencies:
  -> python-dlib
  -> python-face_recognition
==> Checking buildtime dependencies...
==> ERROR: Could not resolve all dependencies.
 -> error making: howdy-exit status 8
 -> Failed to install the following packages. Manual intervention is required:
python-dlib - exit status 4
python-face_recognition - exit status 8
howdy - exit status 8

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable):Arch linux

Howdy version (`sudo howdy version`):

JediBrooker, 2024-01-01:
+1

`==> Starting prepare()...
bsdtar: Removing leading '/' from member names
patching file Makefile.pre.in
patching file Lib/tarfile.py
patching file Misc/NEWS.d/next/Library/2020-07-12-22-16-58.bpo-39017.x3Cg-9.rst
patching file Lib/test/test_urllib2.py
patching file Lib/urllib2.py
patching file Lib/httplib.py
patching file Lib/test/test_httplib.py
patching file Lib/plistlib.py
patching file Lib/test/test_plistlib.py
patching file Misc/NEWS.d/next/Security/2020-10-19-10-56-27.bpo-42051.EU_B7u.rst
patching file Lib/test/multibytecodec_support.py
patching file Misc/NEWS.d/next/Tests/2020-10-05-17-43-46.bpo-41944.rf1dYb.rst
patching file Misc/NEWS.d/next/Security/2020-05-28-06-06-47.bpo-40791.QGZClX.rst
patching file Modules/operator.c
patching file Lib/ctypes/test/test_parameters.py
patching file Misc/NEWS.d/next/Security/2021-01-18-09-27-31.bpo-42938.4Zn4Mp.rst
patching file Modules/_ctypes/callproc.c
patching file Doc/library/cgi.rst
patching file Doc/library/urlparse.rst
patching file Lib/cgi.py
patching file Lib/test/test_cgi.py
patching file Lib/test/test_urlparse.py
patching file Lib/urlparse.py
patching file Misc/NEWS.d/next/Security/2021-02-14-15-59-16.bpo-42967.YApqDS.rst
patching file Lib/test/multibytecodec_support.py
patching file Lib/test/test_minidom.py
patching file Lib/test/test_xml_etree.py
==> Sources are ready.
==> Making package: python2 2.7.18-8 (Mon 01 Jan 2024 18:55:00)
==> Checking runtime dependencies...
==> Checking buildtime dependencies...
==> WARNING: Using existing $srcdir/ tree
==> Removing existing $pkgdir/ directory...
==> Starting build()...
checking build system type... x86_64-pc-linux-gnu
checking host system type... x86_64-pc-linux-gnu
checking for python2.7... no
checking for python3... python3
checking for --enable-universalsdk... no
checking for --with-universal-archs... no
checking MACHDEP... linux2
checking EXTRAPLATDIR... 
checking for --without-gcc... no
checking for --with-icc... no
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... configure: error: in `/home/cbrooker/.cache/yay/python2/src/Python-2.7.18':
configure: error: cannot compute suffix of executables: cannot compile and link
See `config.log' for more details
==> ERROR: A failure occurred in build().
    Aborting...
 -> error making: python2-exit status 4
 -> Failed to install the following packages. Manual intervention is required:
python2 - exit status 4
`

Reginald-Gillespie, 2024-02-15:
I was able to install it from source cloning the repo and running the meson commands although I had to install dlib and the other package system-wide since howdy runs as sudo, previously I only had them installed as user. (They're still building right now but it looks like it should work since I previously did the same thing only for user-wide)

abdul-kaium5, 2024-02-15:
it was auto solved idk anything !

closed 2024-02-15 by abdul-kaium5

nursoda, 2024-02-29:
Had the same error on Arch (EndeavourOS) using [howdy-beta-git](https://aur.archlinux.org/packages/howdy-beta-git) from AUR and no idea how to fix it. Could anyone explain what that cpython build error message "lto1: fatal error: bytecode stream in file ‘dlib_build/libdlib.a’ generated with LTO version 12.0 instead of the expected 13.0
compilation terminated." actually means? Is this an Arch (AUR) issue with the [PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=howdy-beta-git)? What needs to be changed there?

nicojx, 2024-03-12:
Also see this comment and the following ones (i.e. those above): https://aur.archlinux.org/packages/python-dlib#comment-956388

Apparently `pacman` now makes option link time optimization (LTO) enabled by default.

And you might be able to fix it  by uncommenting the `##options=(!lto)` in file PKGBUILD.

Reginald-Gillespie, 2024-03-13:
Good find - I don't want to test it in case I break my current install (although I probably will at some point for the sake of being able to update howdy with pacman), but yes the error mentioned there, "generated with LTO version 12.0 instead of the expected 13.0", is the same one preventing me from installing dlib from pacman iirc so most likely that would resolve the issue.