boltgolt/howdy #875: not being prompted in beta

JediBrooker, 2024-01-02:
I just installed the beta version from the AUR (i can't install v2 because python2 has been removed from arch and it won't install from the AUR either for some reason that I'm not willing to investigate).

However, I'm not being prompted for face unlock in any situation.

"sudo howdy add" and "sudo howdy test" works fine and is using the correct IR sensor on my webcam.

I've added: auth sufficient /lib/security/pam_howdy.so to my appropriate pam.d files (sddm, sudo, kde)

I've looked at this thread issue 785 but I don't think it's relevant because it refers to python2:
https://github.com/boltgolt/howdy/issues/785#issuecomment-1533002545

----

I've searched for similar issues already, and my issue has not been reported yet.

Linux distribution (if applicable): EndeavourOS

Howdy version (`sudo howdy version`): 3.0.0 BETA

xXPerditorXx, 2024-01-07:
I have the same problem too, but mine is about 1 month not working.

musikid, 2024-04-15:
What does `journalctl -xe -t pam_howdy` give?

xXPerditorXx, 2024-04-15:
It just tells `— No entries —`

joehillen, 2024-05-02:
I get:

```
May 02 11:29:35 shadow pam_howdy[823038]: Failed to read files from glob: 3
May 02 11:29:35 shadow pam_howdy[823038]: Underlying error: No such file or directory (2)
May 02 11:29:36 shadow pam_howdy[823038]: pam_unix(i3lock:auth): authentication failure; logname=joe uid=1000 euid=1000 tty=:0 ruser= rhost=  user=joe
```

joehillen, 2024-05-02:
```
glob("/proc/acpi/button/lid/*/state", 0, nullptr, &glob_result);
```
It's looking for a lid, which I don't have because this is a desktop.

joehillen, 2024-05-02:
It's a red herring. I think that's an error that showed up because I copied over the old `/usr/lib/security/howdy/config.ini` to `/etc/howdy/config.ini`

joehillen, 2024-05-02:
It turns out the problem was that all my old models were gone after upgrading from AUR `howdy` to `howdy-git`, so my issue is unrelated.