boltgolt/howdy #884: IR emitter doesn't turn on/ stops if room is too dark.

RickMarijn, 2024-02-03:
When starting 'sudo howdy test' in a well lit room the IR emitter turns on and stays on.
When starting 'sudo howdy test' in a pitch black room the IR emitter doesn't turn on.
When starting 'sudo howdy test' in a well lit room and turning the lights off afterwards the IR emitter goes from on to off.
I have no clue as to why this is happening.

----

I've searched for similar issues already, and my issue has not been reported yet.
I have found one about the IR emitter turning on for a split second, but my issue isn't described perfectly.
Linux distribution (if applicable):
Fedora
Howdy version (`sudo howdy version`):
2.6.1

RickMarijn, 2024-02-04:
Just figured out it only happens after having suspended the laptop.

DIDIK3V1, 2024-04-18:
I have a similar problem, sometimes the ir emitter blinks red sometimes it does not no clue at what is happening, when it blinks howdy is instant, when it does not takes forever and often time out. Using mpv to visualize the ir webcam yield the same result