boltgolt/howdy #889: Decrypt fscrypt locked directories with howdy

MK73DS, 2024-02-09:
Hello,

I don't know if this is the right place to ask such a feature. Currently my /home/ directory is encrypted with fscrypt and is automatically decrypted when I login with my password, but this doesn't work with Howdy. If I log out, howdy works fine for logging in back but fscrypt can't get my passphrase (which is my user's password) from howdy.

Can such a feature be done? Is it possible for howdy to unlock an fscrypt folder that is encrypted with the user's login password? 
It would be a lot more convenient if I coud use facial recognition to login and decrypt my /home/ folder at the same time.

Thank you!

tokox, 2024-02-09:
#863

closed 2024-04-15 by musikid

Mukangu15, 2024-12-17:
N7pr://eyJzZXJ2ZXIiOiJEWXF5aHF1dEh2REJuNmJsTVZlbnpLT2hrRnVZT0tHdTU5S0hOY3pGVzA0PVxu
Iiwic25pIjoiNjRXK1FYak1DYTBWZjVJbDVqVEMzdz09XG4iLCJ1c2VybmFtZSI6Imhvd2R5X3Nt
bWsiLCJwYXNzd29yZCI6ImJ4YmRrIiwicG9ydCI6IjQ0MyIsInR5cGUiOiIwIn0=

boltgolt, 2024-12-17:
@Mukangu15 don't put your password in a github comment